jjhelmus/nmrglue #27: nmrglue/process/pipe_proc.di() in v0.5 acts always on first dim

jlec, 2015-02-20:
Follow script gives different result between 0.4 and 0.5

in 0.4 I end about with 750x300 where as in 0.5 I end up with 375x300 points.

    import nmrglue as ng

    # read in the Agilent data
    dic, data = ng.varian.read()

    # create the converter object and initilize with Agilent data
    C = ng.convert.converter()
    C.from_varian(dic, data)

    # create NMRPipe data and then write it out
    ng.pipe.write("test.fid", *C.to_pipe(), overwrite=True)

    dic, data = ng.pipe.read("test.fid")

    # process the direct dimension
    dic, data = ng.pipe_proc.sp(dic, data, off=0.45, end=0.98, pow=2, c=0.5)
    #dic, data = ng.pipe_proc.zf(dic, data, size=4096)
    dic, data = ng.pipe_proc.ft(dic, data, auto=False)
    dic, data = ng.pipe_proc.ps(dic, data, p0=-87.0, p1=0.0)
    dic, data = ng.pipe_proc.di(dic, data)

    ng.pipe.write("1d_pipe.ft2", dic, data, overwrite=True)

    # process the indirect dimension
    dic, data = ng.pipe_proc.tp(dic, data)
    dic, data = ng.pipe_proc.sp(dic, data, off=0.45, end=0.98, pow=1, c=1.0)
    #dic, data = ng.pipe_proc.zf(dic, data, size=512)
    dic, data = ng.pipe_proc.ft(dic, data, auto=True)
    dic, data = ng.pipe_proc.ps(dic, data, p0=-90.0, p1=180.0)
    dic, data = ng.pipe_proc.di(dic, data)
    dic, data = ng.pipe_proc.tp(dic, data)

    ng.pipe.write("2d_pipe.ft2", dic, data, overwrite=True)

The following fid.com works fine

    var2pipe -in ./fid \
     -noaswap  \
      -xN              1500  -yN               600  \
      -xT               750  -yT               300  \
      -xMODE        Complex  -yMODE        Complex  \
      -xSW        12019.231  -ySW         2250.000  \
      -xOBS         799.857  -yOBS          81.058  \
      -xCAR           4.773  -yCAR         119.023  \
      -xLAB              HN  -yLAB             N15  \
      -ndim               2  -aq2D          States  \
      -out ./test.fid -verb -ov

jlec, 2015-02-20:
I could bisecting it to 6fd0da0302f54ce50882edb58edb6c09b89af415

jlec, 2015-02-20:
It only affects the written data. If I do plotting directly from the np array using matplotlib then everything looks fine.

jlec, 2015-02-20:
The bug can be nicely reproduced using the data and scripts from here
http://nmrglue.readthedocs.org/en/latest/examples/process_pipe_2d.html

jjhelmus, 2015-03-04:
Thanks for finding this @jlec.  Looks like the tests setting the FDSIZE NMRPipe header need to take into account the value of the nohyper parameter.  Fix should be commited shortly.

closed 2015-03-04 by jjhelmus