jjhelmus/nmrglue #28: Scaling of data using bruker.read_pdata

atomman, 2015-04-24:
I have noticed that when I read in a spectrum using bruker.read_pdata the resulting spectrum has a different scaling compared to the raw data viewed in Topspin (or imported into matlab)

I'm on version 0.5-dev.

Two spectra imported with bruker.read_pdata
[image]

Raw data viewed in Topspin
[image]

notice the different scale for the peak (ca 100k in topspin) and >200k when imported with nmrglue.

code:
for i,j in enumerate(expno):

    pathEnd='/%d/pdata/%d' % (j,1)
    pathFinal=path+pathEnd
    dic, data = ng.bruker.read_pdata(pathFinal)
    SF=dic['procs']['SF']
    SW_p=dic['procs']['SW_p']
    SI=dic['procs']['SI']
    OFFSET=dic['procs']['OFFSET']
    ppm_scale=np.linspace(OFFSET,OFFSET-SW_p/SF,SI)
    length = data.shape[0]

    ## Read data into array
    if i==0:
        spec = np.empty(shape=(length,np.size(expno)))
        ax = np.empty(shape=(length,np.size(expno)))
    spec[:,i]=data
    ax[:,i]=ppm_scale

atomman, 2015-04-24:
Note also how the two datasets are scalede relatively different when imported with Nmrglue, which is not the case in the topspin representation of the data

jjhelmus, 2015-04-24:
nmrglue reads Bruker data from fid, ser, 1rr, etc files and performs no scaling of the data.  The data array that is returned from a `nmrglue.bruker.read_` function is the same as the values which are stored in the files. The only modification that nmrglue performs on this data is to recast the data to a complex data type if the last dimension is complex.  This transformation is not performed by the `nmrglue.bruker.read_pdata` function.  

Your plot of the data read using nmrglue corresponds to the data as it is stored in the file.  I believe Topspin scales this data internally when representing the data in a plot which is why you are seeing different results.  I am not familiar enough with Topspin to know how this scale factor is calculated, nor do I have access to Topspin to attempt to determine its internal scale factor.  You might try asking a similar question on the [nmrglue mailing list](http://groups.google.com/group/nmrglue-discus). That lists includes a number of nmrglue users who work with Bruker data and may know how to scale the data in the pdata files to match that in Topspin.

atomman, 2015-04-27:
Hi,
Yes you are right. There is a parameter in the procs file called NC_proc which is a scaling factor for the data.

Data = Data/(2^-Procs.NC_proc); (source rbnmr (matlab script))

jjhelmus, 2015-04-27:
Would it be useful to add a `scale_pdata` function to nmrglue that applied this scaling?  I can add such a function to the bruker module quite easily.  I'm apposed to applying this scaling automatically when reading processed Bruker data but have no issue adding the function and even making it an optional extra step when using the `ng.bruker.read_pdata` function

atomman, 2015-04-27:
One argument for having it as an integrated part of read_pdata is that overflow of the 32bit integer could happen after phasing if the raw data was dispersive. A scale_pdata function or a mentioning of the nc_proc parameter in the documentation of read_pdata will help users of Nmrglue

jjhelmus, 2015-04-30:
PR #29 adds a `scale_pdata` function and a **scale_data** parameter to the `read_pdata` function to perform scaling of processed Bruker data in the same manner as is done in Topspin.

closed 2015-04-30 by jjhelmus