jjhelmus/nmrglue #39: Bruker ppm calculation (unit conversion)

mfitzp, 2015-10-13:
It doesn't look like the unit conversion is supported on Bruker format fids, is this correct? When I attempt to use it I get an error that ndims is not in the Bruker dic.

For reference I am able to generate the Bruker ppm scale using the following calculation:

    offset = (float(dic['acqus']['SW']) / 2) - (float(dic['acqus']['O1']) / float(dic['acqus']['BF1']))
    start = float(dic['acqus']['SW']) - offset
    end = -offset
    step = float(dic['acqus']['SW']) / zero_fill_size

    ppms = np.arange(start, end, -step)[:zero_fill_size]

Where `zero_fill_size` is the size of the spectra.

jjhelmus, 2015-10-13:
Unit conversion objects can be created for Bruker files by creating a universal dictionary and using that to create the appropriate unit conversion object.  For example:

``` Python
import nmrglue as ng
dic, data = ng.bruker.read('./bruker_1d/')
udic = ng.bruker.guess_udic(dic, data)
uc = ng.fileiobase.uc_from_udic(udic)
ppm_scale = uc.ppm_scale()
```

What function are you using that gives an error about ndims not being in the Bruker dic?  Some of these features are relatively new, you may need to update your install of nmrglue.  

The parameters you are using to generate a PPM scale are similar to those used by nmrglue in [lines 109-119](https://github.com/jjhelmus/nmrglue/blob/master/nmrglue/fileio/bruker.py#L109-L119) of the Bruker module to populate the universal dictionary.

mfitzp, 2015-10-17:
Quite right I was on an outdated version — I thought this had made it into nmrglue but couldn't figure it out. Thanks!

closed 2015-10-17 by mfitzp