jjhelmus/nmrglue #49: Strip transformed Bruker data is interpreted incorrectly.

kfritzsc, 2016-03-05:
If `bruker.read_pdata` is used to read spectrum that have been striped transformed the `unit_conversion` object is incorrect (assuming the 'unit_conversion` object(s) are made by calling `bruker.guess_udic`  and `fileiobase.uc_from_udic`). 

The bruker parameters to fix this problem are correctly read by `bruker.read_procs_file` and are in the dic generated by `bruker.read_pdata`. There are two parameters for each dimension: STSR and STSI. they can be accessed with dic['procs']['STSR'] and dic['procs']['STSI'] respectively for the first dimension. The other dimensions can be accessed by substituting `procs` with "proc2s", "proc3s" or "proc4s". If the data is not strip transformed STSI is 0 and STSR is equal to the length for the data array in that dimension. 

It would be easy-ish to fix this problem with a special case check in the  'unit_conversion` class but the unit_conversion class is not initiated  with dic but rather udic. It seams silly to modify the definition of a udic for this special case. One could also 'spoof' the acquisition parameters i.e. return a modified sweep width if STSI or STSR is set but these seams like a bad idea. 

The fix for the problem right now is to not strip transform the data, i.e. in TopSpin set STSR and STSI to 0.

If you are wondering why you would want use strip transform data:  On older spectrometers with analog filters it is common to collect a much (!) larger spectrum width in the direct dimension then necessary. It saves space and time to only FT and plot some of the data.

jjhelmus, 2016-03-15:
@kfritzsc I'm certain that you know more about the format of Bruker pdata data that I do.  I never used this type of data while I was doing NMR.  I've never hear of data being "striped transformed", can you give more details on this process.  Is this similar to to oversampling and decimation of data?  

I think the logic of getting the correct `unit_conversion` object from this type of data should lie in the `bruker.guess_udic` function and not the `fileiobase.uc_from_udic`.  The parameters of the universal dictionary should match the data to give the correct unit conversions, even if the file metadata incorrectly identifies these parameters.  

That said I don't complete understand this issue, if you could provide a stand alone example it might be helpful.

kfritzsc, 2016-03-20:
@jjhelmus 

>I've never hear of data being "striped transformed", can you give more details on this process.

Bruker has an option to only save some of the data points after Fourier transform.  The range of point to be saved (i.e. the strip) is set using the options SI, STSI and STSR. SI is the total number of points in a given dimension, to help clarify: the number of points zero filled is SI-TD, where TD is number of points collected. STSI is the 'Strip SI', i.e. the number of point to keep after a FT. STSR is the 'Strip SR', this parameter controls the location of first point to start keeping data (normally from the left of the spectrum).

Graphically:
[image]

In solid-state NMR  it is desirable for many reasons to collect a larger spectral width in the first dimension. On older spectrometers, due to filter restriction it is often the only option. Still it is silly to manipulate all of the extra points. Therefor, STSI and STSR can be used to save a large amount of memory and computation time.

>Is this similar to to oversampling and decimation of data?

It is different. Although we sometimes do something like decimation and some FT tricks in the indirect dimension to prevent zero frequency artifacts. 

>I think the logic of getting the correct unit_conversion object from this type of data should lie in the bruker.guess_udic...

Okay, I will add this functionality to `bruker.guess_udic`.

My current problem is that SI is overwritten by STSI and the set SI is saved as a 'status parameter'. The parameter we need is therefor not in 'procs' but in the 'audit trail' saved to 'auditp.txt', for example '1/pdata/1/auditp.txt' . This file is in a slightly odd JCAMPDX format so I will have to write an additional parser for it.

After I parse the audit file, We should be able to make the 'effective' udic parameters for strip transformed data with something like:

```python
# Careful, untested code.
strip_sw = (SI/sw) * STSI

striip_car = (SI/sw) * STSR + car
```

>That said I don't complete understand this issue, if you could provide a stand alone example it might be helpful.

I will provide an example soon, I can not share the current data I am working on, but can look for some data of a model compound. I will take me a few days to get back to this.

kfritzsc, 2016-03-20:
I forgot about the parameter FTSIZE, that makes things a lot easier.

jjhelmus, 2016-03-23:
Alright, this creates a strip.  Makes complete sense now.  I was thinking in this was time domain data and got lost.  Yes, the best method is to "fake" the various spectral parameters to give the correct parameters to the universal dictionary.  NMRPipe does a similar procedure with the EXT function which extracts a region from a spectrum.  In nmrglue the `ng.pipe_proc.ext` function also adjusts the spectral width and other additional parameters to account for the new layout of the data.  I'll check out #50 shortly.

jjhelmus, 2016-03-23:
I believe this is closed by #49.  @kfritzsc if you can confirm.

kfritzsc, 2016-03-23:
Thank you @jjhelmus!

closed 2016-03-23 by kfritzsc