jjhelmus/nmrglue #51: read function in bruker.py -- floats in shape shows depreciation warning

agrapentin, 2016-05-13:
Hello, 
Since I am yet to learn git ill post this here. I get the following warning when reading a bruker 1d dataset.

```terminal
nmrglue/fileio/bruker.py:1170: VisibleDeprecationWarning: using a non-integer number instead of an integer will result in an error in the future
  return dic, data.reshape(shape)
```

This is fixed by converting to ints (like in guess_shape) after division.

bruker.py:360

before
```python
gshape = tuple(t)
```

after
```python
gshape = tuple([int(flt) for flt in t])
```

jjhelmus, 2016-05-16:
@agrapentin Thanks for finding this bug and reporting it.  The issue is actually that the division a few lines earlier is producing floats but should be using interger division.  This issue pop up a few times in nmrglue and is causing the VisibleDeprecationWarning with NumPy 1.11.  I'll create a fix for these later this week.

agrapentin, 2016-05-17:
Thanks @jjhelmus, seems to be in a few locations as after converting to pipe, ng.pipe.read has the issue as well. 

Integer division. Why didn't I think of that! =D 
Nice work.

PR #52 (2016-05-19): Fix depreciation warnings and other warnings in tests
closes #51
closed 2016-05-19 by jjhelmus