jjhelmus/nmrglue #55: question re proc_lp.lp_1d

mgeorgiadis, 2016-09-16:
I've got something very similar to a 1-D FID in a numpy float32 array, and I was trying to use the proc_lp.lp_1d function to perform a forward linear prediction on the data.  When I run the function, I get the following error attached in the image.  As you can see, I also tried casting the array to a 'complex64' and that didn't fix the problem either.  How do I get around this error?
[image]

closed 2016-09-19 by jjhelmus

mgeorgiadis, 2016-09-19:
Hi,

Thanks for the reply!  As I mentioned in my post, I actually did try to cast it as a complex64, and still got the error, so I'm not sure whats going on there.  If I bypass the bad_root finding routine, I'm able to get it to run.  I know that for NMR, LP is based around it being a complex value, but I've found that LP is also used for speech modeling, which is real-valued.

jjhelmus, 2016-09-19:
> I actually did try to cast it as a complex64

The commented out line you posted did not save the cast values, see the line I proposed.

> but I've found that LP is also used for speech modeling, which is real-valued.

This is true but the setup of the problems are setup very different, I would not expect a NMR based LP routine to be well suited for speech modeling but you are welcome to try.