jjhelmus/nmrglue #56: Autophase not found?

Jmegan042, 2016-10-12:
Hello, 

Perhaps it has been resolved, although I have not found where or how, but the autophase command that according to the docs should have been under ng.proc_autophase.autops does not exist. Has it been moved, disabled, or otherwise abandoned for better options?

jjhelmus, 2016-10-12:
`ng.proc_autophase.autops` should work if nmrglue is imported as ng.  This feature was added to nmrglue after the last release (v0.5) and therefore is not available unless you install nmrglue from the GitHub source.  Have you done this?

Jmegan042, 2016-10-12:
At the punishment of sounding like an idiot, I do not have 0.5. It seems that I installed 0.4, but when I attempted to find the latest version, I can't seem to locate it in the google download page? 

I have also tried to download straight from the github page and attempted to execute the setup.py program. However, It seems to be angry with me. I am getting an error where an exception has occurred. 

"option -f not recognized"

Full Traceback: 

 ---------------------------------------------------------------------------
GetoptError                               Traceback (most recent call last)
//anaconda/lib/python3.5/distutils/fancy_getopt.py in getopt(self, args, object)
    232         try:
--> 233             opts, args = getopt.getopt(args, short_opts, self.long_opts)
    234         except getopt.error as msg:

//anaconda/lib/python3.5/getopt.py in getopt(args, shortopts, longopts)
     94         else:
---> 95             opts, args = do_shorts(opts, args[0][1:], shortopts, args[1:])
     96 

//anaconda/lib/python3.5/getopt.py in do_shorts(opts, optstring, shortopts, args)
    194         opt, optstring = optstring[0], optstring[1:]
--> 195         if short_has_arg(opt, shortopts):
    196             if optstring == '':

//anaconda/lib/python3.5/getopt.py in short_has_arg(opt, shortopts)
    210             return shortopts.startswith(':', i+1)
--> 211     raise GetoptError(_('option -%s not recognized') % opt, opt)
    212 

GetoptError: option -f not recognized

During handling of the above exception, another exception occurred:

DistutilsArgError                         Traceback (most recent call last)
//anaconda/lib/python3.5/distutils/core.py in setup(**attrs)
    133     try:
--> 134         ok = dist.parse_command_line()
    135     except DistutilsArgError as msg:

//anaconda/lib/python3.5/distutils/dist.py in parse_command_line(self)
    463         parser.set_aliases({'licence': 'license'})
--> 464         args = parser.getopt(args=self.script_args, object=self)
    465         option_order = parser.get_option_order()

//anaconda/lib/python3.5/distutils/fancy_getopt.py in getopt(self, args, object)
    234         except getopt.error as msg:
--> 235             raise DistutilsArgError(msg)
    236 

DistutilsArgError: option -f not recognized

During handling of the above exception, another exception occurred:

SystemExit                                Traceback (most recent call last)
<ipython-input-3-e2424d389812> in <module>()
     44         'nmrglue.util'],
     45     package_data={
---> 46         'nmrglue': ['fileio/tests/data/*.f*', 'fileio/tests/data/*.dir/*']},
     47 )
     48 

//anaconda/lib/python3.5/distutils/core.py in setup(**attrs)
    134         ok = dist.parse_command_line()
    135     except DistutilsArgError as msg:
--> 136         raise SystemExit(gen_usage(dist.script_name) + "\nerror: %s" % msg)
    137 
    138     if DEBUG:

SystemExit: usage: __main__.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: __main__.py --help [cmd1 cmd2 ...]
   or: __main__.py --help-commands
   or: __main__.py cmd --help

error: option -f not recognized

jjhelmus, 2016-10-12:
Google code shut down a year or so ago so the links to the 0.5 release is on the [GitHub releases](https://github.com/jjhelmus/nmrglue/releases).  I'll update the INSTALL.txt file to point to the new location.

What command are you using to install from source?  `python setup.py install` should be sufficient.

Jmegan042, 2016-10-12:
Thanks for the help, I was attempting to execute the setup.py, not run the install. Still getting used to it all. 

After this fix, I was able to run the autophase option. Great addition. Thanks for the help.

jjhelmus, 2016-10-13:
Glad I could help.  I updated the links in the `INSTALL.TXT` file with a link to the GitHub releases.

closed 2016-10-13 by jjhelmus