jjhelmus/nmrglue #62: Files not available

saurabh02, 2017-05-09:
Hello,

This package is very useful! Going through your tutorials and examples, all the links to example files (on googlecode) seem to be broken. Are these hosted elsewhere where I can obtain them from?

Thanks!

jjhelmus, 2017-05-09:
The direct links in the example are broken but the files can still be downloaded from the Google Code Archive, https://code.google.com/archive/p/nmrglue/downloads.  You'll need to find the filename you are looking for in that list.  

I would be happy to review a PR that fixes these links or look at an alternative hosting solution for these files.

saurabh02, 2017-05-11:
Thanks a lot for this! 

Sure, I can submit a PR if I end up using this more often. 

Can I ask, I've not been able to read .fid files through nmrglue from sources other than yours. Would that be due to the nature of inconsistencies in format of such files?

jjhelmus, 2017-05-11:
What is the source of the other `.fid` files?  Perhaps the NMRPipe format has changed.  Can you post a sample somewhere?

saurabh02, 2017-05-11:
For example, the files here - http://science.widener.edu/~svanbram/chem465/htmldocs/nmr_unk.html

jjhelmus, 2017-05-11:
As described on that page, those files are in the NUTS format.  Nmrglue does not support that format.

closed 2017-07-28 by jjhelmus