jjhelmus/nmrglue #66: ng.fileio.pipe.read_table broken in Python 3

viochemist, 2017-07-28:
Hi,

I've been playing with this for a while now, but have no idea how to fix it. Using python 3, read_table results in the following when trying to read a 'test.tab'

```
  File "/usr/local/miniconda/py35/lib/python3.5/site-packages/nmrglue/fileio/pipe.py", line 70, in read_table
    raise IOError("%s has no/more than one VARS line" % (filename))
OSError: test.tab has no/more than one VARS line
```

I tracked down line 70 and find that the variable cl is empty. The line:

            if line[:len(k)] == k:

finds no matches (comparing binary and string or something). I adjusted this section to this:

    for bline in f:
        line = bline.decode()
        for k in specials:
            if line[:len(k)] == k:
                cl.append(line)
                break
        else:
            dl.append(line)
    f.close()

and it works better, seems to be making matches, but I get another error later:

```
  File "/usr/local/miniconda/py35/lib/python3.5/site-packages/nmrglue/fileio/pipe.py", line 86, in read_table
    rec = np.recfromtxt(s, dtype=dtd, comments='XXXXXXXXXXX')
  File "/usr/local/miniconda/py35/lib/python3.5/site-packages/numpy/lib/npyio.py", line 1951, in recfromtxt
    output = genfromtxt(fname, **kwargs)
  File "/usr/local/miniconda/py35/lib/python3.5/site-packages/numpy/lib/npyio.py", line 1482, in genfromtxt
    first_values = split_line(first_line)
  File "/usr/local/miniconda/py35/lib/python3.5/site-packages/numpy/lib/_iotools.py", line 219, in _delimited_splitter
    line = line.split(self.comments)[0]
TypeError: Can't convert 'bytes' object to str implicitly
```

From what I can tell, the 'np.recfromtxt' is having trouble with the new StringIO format, and I'm not sure how to correct that part.

viochemist, 2017-07-28:
Here's an example test.tab file

[test.tab.txt](https://github.com/jjhelmus/nmrglue/files/1183951/test.tab.txt)

jjhelmus, 2017-08-03:
Thanks for the example.  I've put in a fix reading and writing of NMRPipe tables in PR #68.

commit df1eb0c0 (2017-08-03): BUG: ng.pipe.read_table works on Python 3

closes #66
closed 2017-08-03 by jjhelmus