jjhelmus/nmrglue #69: Sorry, but still: "Integers not floats should be used for indexing and slicing arrays"

eisenmenger, 2017-08-16:
Although announced to be solved, with numpy 1.9.2, I recently installed via "git clone https://github.com/jjhelmus/nmrglue" (from source).

dic, data = ng.bruker.read_pdata('../Data/3/pdata/1')
print data.shape
print data
thres = data.std() * 2
peakList = ng.analysis.peakpick.pick( data, pthres = thres )

(1024, 2048)
[[ 14099.25  15660.5   15820.25 ...,    441.5   13453.5   17852.25]
 [ 22354.75  24070.25  16389.   ...,   2944.     -547.75   7666.  ]
 [ 17501.5   21159.     3769.25 ...,    866.5  -10845.5   -4526.  ]
 ..., 
 [  1695.5  -15082.25 -16300.75 ...,   1921.    17331.    22866.25]
 [ -6796.25 -16138.25 -11376.75 ...,   4678.75  24169.25  22745.75]
 [  -512.5   -2591.5    1098.5  ...,   2588.5   24818.75  23191.  ]]
Traceback (most recent call last):
  File "findPeaks.py", line 9, in <module>
    peakList = ng.analysis.peakpick.pick( data, pthres = thres )
  File "/usr/lib/python2.7/site-packages/nmrglue/analysis/peakpick.py", line 215, in pick
    ls_classes)
  File "/usr/lib/python2.7/site-packages/nmrglue/analysis/peakpick.py", line 383, in guess_params_slice
    r = extract_1d(region, rlocation, axis)
  File "/usr/lib/python2.7/site-packages/nmrglue/analysis/peakpick.py", line 398, in extract_1d
    return np.atleast_1d(np.squeeze(data[s]))
TypeError: slice indices must be integers or None or have an __index__ method

eisenmenger, 2017-08-16:
Also, with the script from
https://github.com/jjhelmus/nmrglue/wiki/Find-and-fit-peaks-in-a-spectrum

Traceback (most recent call last):
  File "fitPeaks.py", line 25, in <module>                                                                                                                                                                                                   
    peaks = ng.peakpick.pick(data, 25000)                                                                                                                                                                                                    
  File "/usr/lib/python2.7/site-packages/nmrglue/analysis/peakpick.py", line 222, in pick                                                                                                                                                    
    c_ndil)                                                                                                                                                                                                                                  
  File "/usr/lib/python2.7/site-packages/nmrglue/analysis/peakpick.py", line 291, in clusters                                                                                                                                                
    return [labeled_array[i] for i in locations]                                                                                                                                                                                             
IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`) and integer or boolean arrays are valid indices

jjhelmus, 2017-08-16:
@eisenmenger Can you provide the data file that you are looking at which is causing this error?  I tried to replicate the issue with the data from the wiki example and was unable to cause the error.

eisenmenger, 2017-08-16:
Here you are, in 4 parts because of github limits.
[Data4.zip](https://github.com/jjhelmus/nmrglue/files/1228812/Data4.zip)

Concatenate with  cat Data*.zip > data.zip

eisenmenger, 2017-08-16:
next:
[Data3.zip](https://github.com/jjhelmus/nmrglue/files/1228820/Data3.zip)

eisenmenger, 2017-08-16:
next:
[Data2.zip](https://github.com/jjhelmus/nmrglue/files/1228822/Data2.zip)

eisenmenger, 2017-08-16:
& last:
[Data1.zip](https://github.com/jjhelmus/nmrglue/files/1228823/Data1.zip)

eisenmenger, 2017-08-16:
tested to concatenate - did not work out.
Don't know how to provide you with the original Data.zip of 28 MB

jjhelmus, 2017-08-16:
@eisenmenger Can you upload the file to dropbox or a similar file sharing site?

eisenmenger, 2017-08-17:
To send you a download link & password requires an e-mail address of yours.

jjhelmus, 2017-08-17:
jjhelmus [at] gmail work.

jjhelmus, 2017-08-17:
I got the data file and ran the following script without any error:
``` python
import nmrglue as ng
dic, data = ng.bruker.read_pdata('../Data/3/pdata/1')
print data.shape
print data
thres = data.std() * 2
peakList = ng.analysis.peakpick.pick( data, pthres = thres )
```

Can you verify that you have the latest version of nmrglue and provide details about what NumPy and Python version you are using?

eisenmenger, 2017-08-18:
Thank you for your help !

OS: Linux Fedora 23 (64bit)
$ python
Python 2.7.11 (default, Sep 29 2016, 13:33:00) 
[GCC 5.3.1 20160406 (Red Hat 5.3.1-6)] on linux2
>>> import numpy
>>> print( numpy.__version__ )
1.13.0
>>> import nmrglue
>>> print( nmrglue.__version__ )
0.7-dev
>>> import nmrglue as ng
>>> dic, data = ng.bruker.read_pdata('../Data/3/pdata/1')
>>> print data.shape
(1024, 2048)                           
>>> print data
[[ 14099.25  15660.5   15820.25 ...,    441.5   13453.5   17852.25]
 [ 22354.75  24070.25  16389.   ...,   2944.     -547.75   7666. ] 
 [ 17501.5   21159.     3769.25 ...,    866.5  -10845.5   -4526.  ]         
 ...,                                                                                                                                           
 [  1695.5  -15082.25 -16300.75 ...,   1921.    17331.    22866.25]                         
 [ -6796.25 -16138.25 -11376.75 ...,   4678.75  24169.25  22745.75]                      
 [  -512.5   -2591.5    1098.5  ...,   2588.5   24818.75  23191.  ]]                                                               
>>> thres = data.std() * 2                                                                                                                   
>>> peakList = ng.analysis.peakpick.pick( data, pthres = thres )                                                    
Traceback (most recent call last):                                                                                                    
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/nmrglue/analysis/peakpick.py", line 215, in pick
    ls_classes)
  File "/usr/lib/python2.7/site-packages/nmrglue/analysis/peakpick.py", line 383, in guess_params_slice
    r = extract_1d(region, rlocation, axis)
  File "/usr/lib/python2.7/site-packages/nmrglue/analysis/peakpick.py", line 398, in extract_1d
    return np.atleast_1d(np.squeeze(data[s]))
TypeError: slice indices must be integers or None or have an __index__ method

eisenmenger, 2017-08-18:
Same error on a CentOS 7 (64bit) with Python 2.7.5 & numpy 1.12 & nmrglue (0.7_dev, i.e. from git-sources)

jjhelmus, 2017-08-18:
I've tried to replicate this on my own system with the same version of python and numpy without any luck.  The peak picking does not raise an error.

I'm at a bit of a loss on what else to try.  Perhaps the scipy version is causing issue, what version of SciPy is installed. 

Also, what is the output from:

``` Python
import nmrglue as ng
dic, data = ng.bruker.read_pdata('../Data/3/pdata/1')
thres = data.std() * 2
ploc, pseg = ng.analysis.peakpick.find_all_connected(data, thres, True, False)
print(thres)
print(ploc)
print(pseg)
```

`find_all_connected` is being run inside the peak picker to generate the region limits.  If it gives odd results that could explain the error.

eisenmenger, 2017-08-21:
Thank you very much for your patience and support. You're right, with pointing at scipy version !

import nmrglue as ng
dic, data = ng.bruker.read_pdata('../Data/3/pdata/1')
thres = data.std() * 2
ploc, pseg = ng.analysis.peakpick.find_all_connected(data, thres, True, False)

works WITHOUT problems on MAC OSX (python 2.7.12 from Xcode, numpy & scipy 0.17 added by Homebrew), as well as on Fedora 24 (python 2.7.13, numpy 1.11, scipy 0.16.1).

Finally, the problem has been solved by updating 'scipy' (originally vers. 14.1 on Fedora 23) via "pip install --upgrade scipy" (as root) !!

jjhelmus, 2017-08-21:
Great to hear you found a fix.

closed 2017-08-21 by jjhelmus

janEbert, 2022-07-28:
I stumbled upon the same error. For me, downgrading SciPy from 1.8.1 to 1.7.0 also fixed the problem.