jjhelmus/nmrglue #76: Processing 1D Bruker Data, example not working.

Melisius, 2017-11-28:
Hi,

When I try to run the example from http://nmrglue.readthedocs.io/en/latest/examples/proc_bruker_1d.html, it get the following error:

    ---------------------------------------------------------------------------
    TypeError                                 Traceback (most recent call last)
    <ipython-input-1-3cf04936a564> in <module>()
          4 
          5 # read in the bruker formatted data
    ----> 6 dic, data = ng.bruker.read('expnmr_00001_1')
          7 
          8 # remove the digital filter
    
    ~/miniconda3/lib/python3.6/site-packages/nmrglue/fileio/bruker.py in read(dir, bin_file, acqus_files, 
    pprog_file, shape, cplex, big, read_pulseprogram, read_acqus)
        325     # read the binary file
        326     f = os.path.join(dir, bin_file)
    --> 327     null, data = read_binary(f, shape=shape, cplex=cplex, big=big)
        328     return dic, data
        329 
    
    ~/miniconda3/lib/python3.6/site-packages/nmrglue/fileio/bruker.py in read_binary(filename, shape, 
    cplex, big)
        994     # reshape if possible
        995     try:
    --> 996         return dic, data.reshape(shape)
       997 
       998     except ValueError:
   
    TypeError: 'float' object cannot be interpreted as an integer

I installed nmrglue through conda by: conda install -c bioconda nmrglue and I am using Python3.6.

jjhelmus, 2017-11-28:
Can you try updating to the latest release, 0.6?  I believe this was address in #52.

Melisius, 2017-11-28:
Is there a conda channel available that have v. 0.6?

jjhelmus, 2017-11-28:
Not that I know of, `pip install nmrglue` will get you 0.6 and is the recommended install method.

Melisius, 2017-11-28:
Thanks for your quick support

closed 2017-11-28 by Melisius