jjhelmus/nmrglue #80: TopSpin 4.0/Avance Neo Compatibility

sargon14, 2018-05-09:
Hello,

With the release of TopSpin 4.0 and the Avance Neo, Bruker made a change to their data storage format. Data taken in TS4 is now saved as double precision floats (float64) instead of integers (int32). I believe that TS4 is only compatible with Avance Neo spectrometers, so this doesn't matter to very many people yet, but it will probably start to matter going forward (and it affects me already, hence posting this issue). So, the data reading functions in nmrglue.fileio.bruker should probably be updated to allow different data types to be read in (and probably also written out). I'd submit a pull request, but all I've done so far is hacked in a solution (changed "i4" to "f8" in nmrglue.fileio.bruker.get_data), which doesn't seem up to the programming standards of this project!

As for how to make a flexible change, there is now a parameter in acqus that notes what format the fid/ser file is in: 0 means int (the old style), 2 means double. So, I guess the read function should look at that to see what data type it needs to load?

---Robby Blum

jjhelmus, 2018-05-13:
@sargon14 It would be great to add support for the new Bruker double precision format to nmrglue.  If you are interested, I'd be happy to help you add this feature to the library.  I think following the logic used to select the correct endiness of the data, the `BYTEORDA` parameter and `big` variable, might be a good style to mimic.

Do you have a dataset with the new format that you could share?

sargon14, 2018-06-08:
Right, I'm going to pick this back up again, I think. I'll open a pull request as soon as I remember how I'm supposed to do that...