jjhelmus/nmrglue #86: TOPSPIN 4 Incompatible to read processed files

witheve17, 2018-10-10:
Bruker has recently updated there processed file format so when you try to use nmr glue to read a processed file it reds incorrectly the 2rr, and 1r files. I am not sure what changes in the new topspin.

jjhelmus, 2018-10-10:
Can you provide an example of the file in the new format?

witheve17, 2018-10-10:
I have attached the zipped experimental folder 
[189.zip](https://github.com/jjhelmus/nmrglue/files/2464977/189.zip)

jjhelmus, 2018-10-17:
I do not have access to TOPSPIN so I do not know how this data is expected to look.  Plotting the data using the current development version of nmrglue gives a plot that is similar to the on in the `thumb.png` file:

[image]

Do you have any 1D data and a plot that I could compare with?

witheve17, 2018-10-17:
Hi Jonathan,

Thank you for replying. This is how the data is supposed to look. However when I import the data using nmrglue this is what I get for plotting.

This is the code that I use to import the data.

dic, data = ng.bruker.read_pdata('/opt/topspin35pl7/data/Velencia_ABBrushOp/155/pdata/1')
udic = ng.bruker.guess_udic(dic, data)
uc=ng.fileiobase.uc_from_udic(udic)
ppm = uc.ppm_scale()
cl = data.std() * 2 * 1.2 ** np.arange(10)
fig = plt.figure()
ax = fig.add_subplot(111)
ax.contour(data, cl, colors='blue')
plt.show()

This is the resulting graphic.

[/var/folders/yr/7qr6dznx18vgw9w70hvn_dzh001tkf/T/com.microsoft.Outlook/Content.MSO/96ACEECE.tmp]

How ever in topspin there is just one major peak.

Before this I erase nmrglue from my conda nmr processing environment and reinstalled into the environment using conda.
When I evaluate the comman conda list:

nmrglue                   0.5                      py35_0    jjhelmus
is the version that I have,

I am operating Mac Osx HighSierra  I use jupyter notebook as my python interface.

Thank you for your attention to this matter.

Best,
--
Velencia Witherspoon
NRC-NIST Postdoctoral Fellow
Functional Polymers Group
Material Science and Engineering Division
National Institute of Standards and Technology
Email: vjw@nist.gov<mailto:vjw@nist.gov>
Phone: 301-975-4736

[figure_1]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F1050278%2F47057819-ef470800-d187-11e8-8818-6b84d9b9cc5e.png&data=02%7C01%7Cvelencia.witherspoon%40nist.gov%7Ccc8cb273b96e43b7d23508d633d5d877%7C2ab5d82fd8fa4797a93e054655c61dec%7C1%7C0%7C636753390902384825&sdata=4Y%2Bh8h8NtU0c5tvUM6H2d6xC5vcFL5lLgNzuFG1GsaQ%3D&reserved=0>

kaustubhmote, 2018-10-22:
This seems to be related to #81. The change in Topspin 4.0 is the swirch to storing the raw and processed data as floats rather than integers. I can reproduce the issue with nmrglue v0.5 and v0.6, but it seems to work fine with v0.7-dev. 

@witheve17, you will have to download the current development version from github and install it manually to be able to read the data in correctly.

jjhelmus, 2018-10-23:
I agree with @kaustubhmote, upgrading to nmrglue version 0.6 should fix the issue.  If you want to install the development version you can do this using `pip install git+https://github.com/jjhelmus/nmrglue.git`.

witheve17, 2018-10-23:
Hi Thank I have completely uninstalled and re-installed nmrglue and the issue has been fixed.

jjhelmus, 2018-10-23:
Great to hear!

closed 2018-10-23 by jjhelmus