jjhelmus/nmrglue #87: Test datasets: Alternative needed for the Google-Code archive

kaustubhmote, 2018-11-06:
I am planning to push a PR (currently my `master` branch) that writes processed Bruker data back into a format that Topspin can read. This is a fairly big addition and I have included some basic tests to make sure everything works OK. For the new tests to work, I will need to push some additional test datasets (processed topspin data) and some changes to the `conversion scripts`. 

Is there an alternative way of doing this and making them available to the CI, since Google-Code is no longer functional?

jjhelmus, 2018-11-07:
How big of dataset do the tests require? If   For data the ~10-100 kilobyte range including the file directly in the repository in a [data folder](https://github.com/jjhelmus/nmrglue/tree/master/nmrglue/fileio/tests/data).  Larger data sets could be stored in a nmrglue_data repository or as a [release artifact](https://help.github.com/articles/distributing-large-binaries/) which has a 2GB upper limit.

kaustubhmote, 2018-11-08:
The additional datasets are themselves relatively small at ~2.5MB (out of which a single 2D is ~2MB). That is still bigger than the nmrglue codebase (!), but I am testing the ability to read and write processed 2D datasets from and into the bruker format which makes smaller sizes a bit difficult. For the current testing requirements, I will make smaller datasets to fit into the data folder tests in `fileio` and/or change some tests.

However, I think it would be great to have a way to update the [test_data_v0.4-dev.zip](https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/nmrglue/) on google-code archive and run the more extensive tests, atleast locally.

kaustubhmote, 2020-06-19:
@jjhelmus and @JLVarjo I am unable to find the testing data for testing PR #120. the test_data_v04-dev.zip file from above is missing jcampdx datasets. Any idea where I can find these?

JLVarjo, 2020-06-19:
Hi,
I have it and @jjhelmus should have it, but it seems that this test data archive is not updated. I think this data is not more than few megabytes. Should I upload them to the nmrglue_data repo maybe?

kaustubhmote, 2020-06-19:
Maybe you can email it to me for now? We probably should revisit how/where we store these test datasets. It might be best to avoid adding to the repo for now, as most of the datasets will be be larger than than current repo size.

JLVarjo, 2020-06-19:
I agree - it would simplify pushing bug fixes if test data could be made instantly available as well. I'll get back to you next monday and e-mail the data.

JLVarjo, 2024-11-13:
Hi, what's the situation on this, is there an up-to-date test data package available somewhere?

jjhelmus, 2025-02-17:
The latest test data set that I have is included as a release artifact in the 0.5 release, [test_data_v0.5-dev.zip](https://github.com/jjhelmus/nmrglue/releases/download/v0.5/test_data_v0.5-dev.zip). This is old (from 2015) but has the raw data and processing scripts that were used for the tests in the root level `tests` directory at the time.

kaustubhmote, 2025-02-20:
There is quite a bit of new data that has been added over the years and I have managed to collect testing datasets from a few people as well. The complete testing dataset after decompressing and processing required for testing stands at ~5.7G. I will try and compress this down, and also see that everything passes (there are several failing tests currently) and then try and figure out a way to share this.  

Another issue is going to be that the example code, especially the ones published in JBNMR needs to be rerun and checked that it still works and produces the same output (which in some cases in unlikely). @jjhelmus do you wish to have a `original` folder that contains the unchanged code for records, or should we rely on the git history?