jjhelmus/nmrglue #97: nmrglue convert function seems to produce unwanted distortions

magnetomeo, 2019-05-23:
Hello everybody,

I recently started my PhD in NMR and discovered nmrglue just a week ago. I was really happy to understand that it envisioned a function, ng.convert.converter(), which allows to translate spectra in different file formats. Because I work with Bruker spectrometer and enjoy analysis with Sparky quite a lot, I am very interested in converting Bruker spectra into .ucsf files.

Now, while the script (below) works, it produces very annoying distortions (along the proton dimension only, apparently) which make the resulting .ucsf spectrum unusable. 

The script is: 

```
import nmrglue as ng

sdic, sdata = ng.bruker.read_pdata('my lovely spectrum in bruker format')
u = ng.bruker.guess_udic(sdic, sdata)

C = ng.convert.converter()
C.from_bruker(sdic, sdata, u)
pdic, pdata = C.to_sparky()

ng.sparky.write("my lovely spectrum in sparky format.ucsf", *C.to_sparky(), overwrite = True)
```
For further reference, I attach two figures of the same spectrum in Sparky format. "distorted" is produced via the above script, while "correct" is yielded through a different script.

[image]
[image]

Apologies if the issue was already tackled somewhere else, I could not find any helpful resource in this respect.

Thanks in advance!

jjhelmus, 2019-05-31:
The convert functionality in nmrglue often requires that the user set the spectral parameters manually.  The values from the `guess_udic` function are guesses, they are not always correct.  From the plots, my guess is that the spectral width of the H dimension is guesses incorrectly.  The [bruker2pip_2d example](https://nmrglue.readthedocs.io/en/latest/examples/bruker2pipe_2d.html) provides an example of how these spectral parameters can be set manually.  Starting from the values in the guessed universal dictionary can give you the other parameters, although these too should be verified.  The tutorial section on [universal dictionaries](https://nmrglue.readthedocs.io/en/latest/tutorial.html#universal-dictionaries) might be helpful as well.

kaustubhmote, 2019-06-08:
@magnetomeo, after you process the spectrum in Topspin, do you save only the downfield part of the spectrum (> ~6ppm) using the 'Save display region to...' and 'Parameters STSR/STSI' options? It does seem like that from the way the 'distorted' spectrum looks.

If yes, you should use `strip_fake=True` while making `udic`, which will give the corrected spectral width and center frequencies. See #49 for more details on how this  #works.
In short, `udic = ng.bruker.guess_udic(dic, data, strip_fake=True)` should work.

magnetomeo, 2019-06-19:
Thanks @kaustubhmote! Your solution worked for me! Indeed, I defined STSR/STSI parameters.
Also, thanks to @jjhelmus for suggesting a more widely applicable fix.

closed 2019-06-19 by magnetomeo