jjhelmus/nmrglue #102: Problem with proc_autophase.autops(data,  "acme")

jakobjakobson13, 2019-07-07:
Dear Jonathan,

i tried your [example](https://nmrglue.readthedocs.io/en/latest/examples/proc_bruker_1d.html) for the automatic phase correction algorithms. However the automatic phasing with the `acme` function

```python3
#! /usr/bin/env python

import nmrglue as ng
import matplotlib.pyplot as plt

# read in the bruker formatted data
dic, data = ng.bruker.read('expnmr_00001_1')

# remove the digital filter
data = ng.bruker.remove_digital_filter(dic, data)

# process the spectrum
data = ng.proc_base.zf_size(data, 32768)    # zero fill to 32768 points
data = ng.proc_base.fft(data)               # Fourier transform
data = ng.proc_autophase.autops(data, "acme")      # phase correction
```

does not work due to the following error message: 
`Warning: Maximum number of function evaluations has been exceeded.`

The `peak_minima` does work fine. So, regardless of this bug in the acme function: Thanks a lot for this package.

Regards

Jakob

jjhelmus, 2019-07-10:
The warning indicates that the autophase algorithm is not converging in a reasonable amount of time.  I do not know enough about the routine to debug this.

jakobjakobson13, 2019-07-10:
Okay, no problem. Should I close the bug report then?

kaustubhmote, 2019-07-11:
This seems like an edge case.
The current `autops` function does not give user acess to `maxiter` and `maxfev` args for the `scipy.optimize.fmin` function, and so by default, these are set to 2*200. Currently `acme` seems to converge in about 350 function evaluations on my computer on this dataset, and thats significantly short of the maximum, but its possible that a different version of scipy/numpy/python will push this over to 400 at which point it will fail. I did manage to make it fail once, but can't seem to repeat that. Changing the start value of p0 helps, and so does the `post_proc=True` method for digital filter removal (#24, #67) to start with a flat basline. This maybe a good time to revisit #48 and make many of these options available.

PR #108 (2019-07-30): Additional options for autops
This PR does 3 things to the `autops` function:

1. Allows an optional `return_phases` argument. If set to `True`, the optimized phases will be returned. By default, this is set to `False`.
2. Allows additional (optional) keyword arguments to be passed to the `scipy.optimize.fmin`. The default options work fine most of the time for `autops`, but this way, one has the option to turn off the (sometimes irritating) convergence messages (using `disp=False`). Also `maxiter` and `ftol` might  be useful in some cases. The complete list of options is documented [here](htt [... cut]
closed 2019-07-30 by jjhelmus