jjhelmus/nmrglue #104: Reading metadata

meh-mah, 2019-07-10:
Hi
I am trying to read bruker data and I need to extract info such as sample name and experiment name. looking at dic generated by bruker.read() and bruker.read_pdata() I could not find such information.
Any suggestion how I can extract these metadata?
Bests
Mehran

jjhelmus, 2019-07-10:
Is this information available in any of the files in the experiment directory?  If not then nmrglue will not be able to determine this information.

kaustubhmote, 2019-07-10:
I think currently these are not being read in, but these are stored at well defined location and simple file reading functions can be used. For example,

    data_loc = "/opt/topspin4.0.7/examdata/exam1d_1H/1"
    dic, data = ng.bruker.read(data_loc)

    sample_info_loc = os.path.join(data_loc, "sample_info.prop")
    title_loc = os.path.join(dataset_location, "pdata", "1", "title")
    
    with open(sample_info_loc) as f:
        sample_info = f.read()

    with open(title_loc) as f:
        title = f.read()

    print(title)
    # 1H Cyclosporin

    print(sample_info) # this can be parsed several different ways
    # Wed Jul 10 21:45:48 IST 2019
    # I_10000=NM=Sample ID, NM2=Example File, HEIGHT=0
    # I_10001=NM=Origin, NM2=A, HEIGHT=0
    # I_10002=NM=Chemical formula, NM2=B, HEIGHT=0
    # I_10003=NM=Concentration, NM2=C, HEIGHT=0
    # I_10004=NM=Date Prepared, NM2=D, HEIGHT=0
    # I_10005=NM=Buffer, NM2=E, HEIGHT=0
    # I_10006=NM=Contact, NM2=F, HEIGHT=0
    # I_10007=NM=Comment, NM2=G, HEIGHT=5
    
Maybe a couple of additions to `bruker.read` and `bruker.read_pdata` can do this default, just to avoid the boilerplate?

meh-mah, 2019-07-20:
Thanks a lot for clarification. Actually it seems that there is no file holding such metadata.