jjhelmus/nmrglue #109: write 2D Bruker processed data

gul916, 2019-09-25:
Hello and thank you for all your work. 
I have a problem to save a 2D Bruker processed data file. Here is my code, used under Linux CentOS 7.7 with Python Anaconda 3.7, Spyder 3.3.6 and nmrglue 0.7.
Could you please help me to solve this ?
Thank you by advance.

```
In [1]: import nmrglue as ng
     ...: ng.__version__
Out[1]: '0.7'
In [2]: data_dir = '2D/1/pdata/1'
     ...: dic, data = ng.bruker.read(data_dir)
     ...: ng.bruker.write_pdata(data_dir, dic, data, overwrite=True)
Traceback (most recent call last):

  File "<ipython-input-17-f7a36b4de334>", line 3, in <module>
    ng.bruker.write_pdata(data_dir, dic, data, overwrite=True)

  File "/opt/anaconda3/lib/python3.7/site-packages/nmrglue/fileio/bruker.py", line 891, in write_pdata
    data = reorder_submatrix(data, shape, submatrix_shape, reverse=True)

  File "/opt/anaconda3/lib/python3.7/site-packages/nmrglue/fileio/bruker.py", line 1462, in reorder_submatrix
    return rdata.reshape(shape)

**ValueError: cannot reshape array of size 0 into shape (64,512)**
```

kaustubhmote, 2019-09-26:
Hi, based on the error stack, it seems that the array `data` is empty. 
When you are reading in the data, can you try with `ng.bruker.read_pdata` instead of `ng.bruker.read`. Your directory seems to be a processed data directory, but you seem to also have raw data in it, since the `ng.bruker.read` line does not raise any error. If this does not work, you should look at whether `data` is still empty, and if not, please post the output of `data.shape` for further debugging.

Jmegan042, 2019-09-26:
Hey everyone, while we are discussing bruker 2D data...

Is anyone able to use NUS 2D data? Is there a way to port over the reconstruction that topspin uses (or even invoke it from a command line through subprocess)?

I can run it through topspin, then render, but if I could remove topspin from my processing scheme, that would be awesome.

Any ideas?

gul916, 2019-09-26:
@kaustubhmote 
Importation works perfectly. I can process, plot the data or whatever. The data set is not empty and has the expected shape. The advantage to use the processed directory when importing with ng.bruker.read is that you can directly use the same directory with ng.bruker.write_pdata. The problem is the same if I use read with the expno directory and write_pdata with the procno directory.
However, if I use read_pdata and write_pdata, it works, but I want to use raw data.

@Jmegan042 
For Non-Uniform Sampling (NUS), you can use [MddNMR](http://mddnmr.spektrino.com/) that is working under Linux or MAC with NMRpipe. You can then import data to python with nmrglue, which I tested. I suppose a full automated workflow can be implemented by importing Topspin data, converting it to nmrpipe, processing it with mddnmr, importing back nmrpipe results and converting it to bruker, but I have not yet tested.

kaustubhmote, 2019-09-26:
@gul916, I see, the problem seems to be in the `reorder_submatrix` routine which does not handle raw data correctly. I am guessing that if correct values are given for the `submatrix_shape` parameter in the `write_pdata` function, and the appropriate parameters are changed in `procs` and `proc2s` file (so that they match the raw data), this will work. What those values should may take a bit of time to figure out, but the following hack works:
In Topspin, process the data without using and additional zero-filling (keep SI in both the dimensions equal to TD for that dimension). Read the raw data so that the `procs` files that are read belong to the zero-fill-less data (this currently requires that pdata folder is 1). Writing back to another processed data using this new dictionary works fine. I am able to get Topspin to read raw data as processed data using this. In general, getting topspin to write out the procs file with appropriate dimensions works with either raw or processed data. But, this is just a hack, and I am sure editing the files in appropriate places so that they are consistent with the data being written out should work via nmrglue.

gul916, 2019-09-26:
@kaustubhmote 
With your indications, I was able to make `write_pdata` work with the following code, whatever the zero-filling in procno.
Thanks a lot.

```
data_dir = '2D/1/pdata/1'
dic, data = ng.bruker.read(data_dir)
data = ng.proc_base.zf_size(data, dic['procs']['SI'])
data = ng.proc_base.tp_hyper(data)
data = ng.proc_base.zf_size(data, dic['proc2s']['SI'])
data = ng.proc_base.tp_hyper(data)
ng.bruker.write_pdata(data_dir, dic, data, overwrite=True)
```

closed 2022-07-22 by kaustubhmote