jjhelmus/nmrglue #110: Reading and vizualizing .rda nmr spectroscopy files

alexandrova-s, 2019-11-28:
Hi,
I am writing my thesis and I have to analyze the spectrum from Siemens apparatus in .rda format. I would love to use nmrglue for analysis, because I really like the methods of this module.
Unfortunately, I have a problem loading my specters. Is there a method in nmrglue that will allow me to load my .rda data and it will be ready for further analysis and visualization?
I really need leftshift and line broadening in my spectra, because they are slightly shifted relative to each other, and I would like to present them on a common graph.
I am asking for help very much.
I have already tried to load the spectrum as follows:

dic, data = ng.pipe.read_2D(".\whoiii3\whoiii_30_2.rda")
print(data.shape)
print(data.dtype)

but it gives me answer like that:

C:\Users\Aleksandra\AppData\Local\Programs\Python\Python37\lib\site-packages\nmrglue\fileio\pipe.py:1478: UserWarning: (4103,)cannot be shaped into(0, 0)
  warn(str(data.shape) + "cannot be shaped into" + str(shape))
(4103,)
float32

and such results are not possible for further analysis.

thank you in advance for your help :)

kaustubhmote, 2019-12-01:
Nmrglue currently does not read this format automatically. The command you are using is applicable for nmrpipe data alone. You may have to manually read in your dataset as a numpy array (see [numpy.load](https://docs.scipy.org/doc/numpy/reference/generated/numpy.load.html), [numpy.loadtxt](https://docs.scipy.org/doc/numpy/reference/generated/numpy.loadtxt.html) and [numpy.fromfile](https://docs.scipy.org/doc/numpy/reference/generated/numpy.fromfile.html)), and then manually make a [universal dictionary](https://nmrglue.readthedocs.io/en/latest/tutorial.html?highlight=universal%20dictionary). Then you should be able to use the nmrglue processing functions. 

It may help if you post an example dataset, if someone would like to make a PR to add this functionality to nmrglue at some stage.

jjhelmus, 2019-12-02:
The [documentation for the suspect](https://suspect.readthedocs.io/en/latest/notebooks/tut01_intro.html) indicates that it may be able to read in Siemens rda files.  The library is [open source](https://github.com/openmrslab/suspect), it might be useful to study if anyone wants to add this functionality to nmrglue.

alexandrova-s, 2019-12-05:
Thank you both very, very much for your help :)
I hope to manage it