jjhelmus/nmrglue #112: Thermo/PicoSpin 80 JDX read

sjlukacs, 2019-12-19:
hello one and all,

I'm trying to read JDX files from a thermo fisher, aka, picoSpin, Model 80 Series II H-NMR into python.  any ideas on how to do that?  below is the ascii dump of acetonitrile.jdx file.  thank you in advance and have a great day.

Lucas

##TITLE=            onePulse-2019-12-17 17:58:22
##JCAMP-DX=         6.0
##DATA TYPE=        NMR FID
##DATA CLASS=       NTUPLES
##ORIGIN=           picoSpin
##OWNER=            My Name
##LONG DATE=        2019/12/17 17:58:22+0000
##.OBSERVE FREQUENCY=82.0177140462
##.OBSERVE NUCLEUS= ^1H
##.ACQUISITION MODE=SIMULTANEOUS
##.DELAY=           (12, 12)
##SPECTROMETER/DATA SYSTEM=picoSpin-45
##$PS_CREATORINSTITUTION=Thermo Fisher Scientific
##$PS_DESCRIPTION=One pulse NMR script where the pulse is followed by a receiver
recovery delay before acquiring the specified number of acquisition points. If
the number of scans > 1 the script attempts to wait for the recycle delay before
the next pulse. If 'align-avg. data' is set, the data sent to the plot window
will be the average of the  data acquired so far. The data is averaged only
after being spectrally aligned using a FT-based method. The quality of the
alignment goes up with the number of 'zero filling' which should be a power of
two for speed.
##$PS_CREATOR=
##$PS_TITLE=        onePulse
##$PS_HARDWARE=     80
##$PS_VERSION=      1.3
##$PS_CREATOREMAIL= erick.winston@thermofisher.com
##$PS_MODIFICATIONDATE=2000\/01\/01 00:08:12
##$PS_SHORTNAME=    one pulse
##$PS_CREATIONDATE= 2011\/03\/03
##$PS_SEQGRAPHIC=   pulse_acquisition.svg
##$PS_JCAMP AVG.=   True
##$PS_JCAMP IND.=   False
##$PS_ZERO FILLING= 8192
##$PS_ALIGN-AVG. DATA=True
##$PS_AUTO PHASE=   True
##$PS_PHASE CORRECTION=-8.1 degrees
##$PS_EXP. APODIZATION=0 Hz
##$PS_TX FREQUENCY= 82.0177140462 MHz
##$PS_AUTO TX=      True
##$PS_AUTO TX OFFSET=200 Hz
##$PS_REQUESTED SCANS=10
##$PS_COMPLETED SCANS=10
##$PS_PULSE LENGTH= 45 us
##$PS_ACQUISITION POINTS=3072
##$PS_AUTO RCVR TIMING=True
##$PS_RECOVERY DELAY=5.0 us
##$PS_RECYCLE DELAY=8 s
##$PS_BANDWIDTH=    4 kHz
##$PS_POST FILTER ATTENUATION=auto
##$PS_MAX TIME TO PLOT=1000 ms
##$PS_MIN FREQ. TO PLOT=-2000 Hz
##$PS_MAX FREQ. TO PLOT=2000 Hz
##$PS_MAX PLOT POINTS=4096
##$PS_LIVE PLOT=    True
##.SHIFT REFERENCE= INTERNAL,unknown,1536,0.000
##NPOINTS=          3072
##NTUPLES=          NMR FID
##VAR_NAME=               TIME,   FID/REAL,   FID/IMAG, PAGE NUMBER
##SYMBOL=                    X,          R,          I,          N
##VAR_TYPE=         INDEPENDENT,  DEPENDENT,  DEPENDENT,       PAGE
##VAR_FORM=               AFFN,       ASDF,       ASDF,       AFFN
##VAR_DIM=                3072,       3072,       3072,          2
##UNITS=               SECONDS, ARBITRARY UNITS, ARBITRARY UNITS,
##FIRST=                     0,     659519,     349220,          1
[... 475 lines cut ...]
0.6882500052B0590A1792F203a120h094a5029b2936a9512b6266b5241a7775a6986e057B460
0.69174999A0905A6536B0394B6857B6086B2567A8403A2273E143e116a2110a7410b2130b2937
0.6952499747b4179b0594a8948a3154h386D876A1290A6374A9205B5974B3511B3542A5899
0.6984999776A2666D308c22i268a6561a9603b3246b2875b0869a5022a3022f364D409A1426
0.7020000219A5138A8808B3331B3120B1951A9130A2349D941a342i779a6394b1323b4315b5340
0.7055000067b3130a7296a0935c224B754A0047A6181A8159B5986B1021B2179A4055I844D502
0.7089999914c158a1190a6118b0069b5059b5643a9801a7290a0724a534A866A0587A5072B0708
0.7124999762B2330B5170A9900A5233A1074E227e473a1218a6529b0995b2247b2550b0818
0.715749979a6722a2486c873E111I933B1085B2812B2708B2987B0405A4635I316A046b952
0.7192500234a2732a6756b1835b1827b2341a7291a8596h406a960G917A2234A9965A9717B4688
0.7227500081B1878A9216A3896G655h66d603a2116b0433b1805b2508b3340a9145a3498f419
0.7262499928a073D916G305B0808A9203B6445B1198A9018A1056D184A179d764a0326a9391
0.7297499776a7316b2389b0077a6892a4564e522C122F203A0958A8104B0942B1632A8984A9039
0.7332500219A4747F246a369f600a3678a7431a5981b0646a8750a5789a2345d444A180D083
0.7367500067A2357A7101B1169A6971A8736A9861A4493D093h31f296a2634a6372a8779b0701
0.7402499914a7785a6046a3487e569A157G690A3550A8458B0516B1539A9249A6095A2088G107
0.7437499762c397e271a1675a6227a9919b0861a9934a7890a3028f817A712C741A2978A6332
0.7472500205B2066A9696A7803A4052A4502H14c42h583g172a9694b0464a9653b0883a6366
0.7507500052a2391b987B027F760A2593A5772A7104A9452A4956A5913I550C062a577i712
0.75424999a2758a7452a8832a9523a9352a6792i064c060C121A1176A3772A8517A8980B1481
0.7577499747A6344A4367A0269B757d754f388a7742a9666b0630b0512a8634a2948i188g86
0.7612500191C066A1203A5263A8721B0160A9078A6444A0863F485B003d987i123a3065a6162
0.7647500038b2588b0919a4979a6013g174b118E414G846A1236A7741A9639A7261A6381
##END TUPLES=       NMR FID
##END=

kaustubhmote, 2019-12-19:
This seems to work (with a minor warning):

    import  nmrglue as ng
    dic, data = nmrglue.jcampdx.read(data)

The warning is only because $PS_CREATOR is not given in your file.

sjlukacs, 2019-12-19:
that is cool, I believe that worked.  under data, it returned a python list of two numpy.ndarray s.  I'm assuming that the two are real and imaginary data as shown above in my text dump under ##DATA-TABLE ...  XYDATA s.  do you think I correct there?  do you have any suggests on turning it into a ppm spectrum?

kaustubhmote, 2019-12-20:
Yes, the two arrays are the real and the imaginary parts of the FID.

1. For processing, you can follow the standard instruction set given [here](https://nmrglue.readthedocs.io/en/latest/examples/proc_bruker_1d.html). This is for Bruker datasets, so ignore the `remove_digital_filter` part, and the rest can be used as is.
 
2. For getting the [universal dictionary](https://nmrglue.readthedocs.io/en/latest/tutorial.html#universal-dictionaries) for your data, you can use, `ng.jcampdx.guess_udic(dic, data)` and use it as explained [here](https://github.com/jjhelmus/nmrglue/issues/39#issuecomment-147805096).

jjhelmus, 2019-12-20:
Here is a script that might get you started.  I determined the phase correction using the [nmrglue.process.proc_autophase.manual_ps](https://nmrglue.readthedocs.io/en/latest/reference/generated/nmrglue.process.proc_autophase.manual_ps.html#nmrglue-process-proc-autophase-manual-ps) function.  

``` Python
import matplotlib.pyplot as plt
import numpy as np
import nmrglue as ng

# read in the data
fname = "picospin.jcamp"
dic, raw_data = ng.jcampdx.read(fname)

# create complex array
npoints = int(dic["NPOINTS"][0])
data = np.empty((npoints, ), dtype='complex128')
data.real = raw_data[0][:]
data.imag = raw_data[1][:]

# process
data = ng.proc_base.zf_size(data, 8192)
data = ng.proc_base.fft(data)
data = ng.proc_base.ps(data, p0=-27.7, p1=0.0)
data = ng.proc_base.di(data)

# determine the ppm scale
udic = ng.jcampdx.guess_udic(dic, data)
udic[0]['car'] = 0.0
uc = ng.fileiobase.uc_from_udic(udic)
ppm_scale = uc.ppm_scale()

# plot
fig = plt.figure()
ax = fig.add_subplot(111)
ax.plot(ppm_scale, data)
plt.show()
```
[image]

The scale looks off to me, some of the parameters in the universal dictionary are likely determined incorrectly and need to be set manually.

sjlukacs, 2020-01-24:
yes, that code really got me moving in the right direction.  thank you so so much.  i was able to fix the ppm scale as shown in the final below code.  the ppm scale for the same file are shifted only 0.03 as compared to the mnova software.

def convertNMR_RawPicoSpin80_to_ndarray(txt):
    #where txt is the raw text of the full .jdx file.
    tfile = tempfile.mkstemp(prefix="nmr_", suffix=".jdx")[1]
    ffile = open(tfile, 'w')
    ffile.write(txt)
    ffile.close()
    dct, raw = ng.jcampdx.read(tfile)
    data = numpy.empty(len(raw[0]), dtype='complex128')
    data.real = raw[0]; data.imag = raw[1]
    #process fid (time-domain) to spectrum (frequency-domain)
    data = ng.proc_base.zf_size(data, int(dct['$PSZEROFILLING'][0])) #8192
    data = ng.proc_base.fft(data)
    data = ng.proc_base.ps(data, p0=float(dct['$PSPHASECORRECTION'][0].replace(' degrees', '')), p1=0.0)
    #data = ng.proc_base.di(data)
    data = numpy.absolute(data)
    #determine the ppm scale
    udct = ng.jcampdx.guess_udic(dct, data)
    udct[0]['car'] = 0.
    udct[0]['sw'] = float(dct['$PSMAXFREQ.TOPLOT'][0].replace(' Hz', '')) - float(dct['$PSMINFREQ.TOPLOT'][0].replace(' Hz', '')) #4000. #3985.
    uc = ng.fileiobase.uc_from_udic(udct)
    return uc.ppm_scale(), data

JLVarjo, 2020-02-13:
Hi, commenting here since I'm the one who wrote the JDX parser. To be honest, I didn't put much effort to JDX FIDs since I usually deal with already processed data. Apparently, at least the guess_udic should behave differently in the case of FID input, when calculating the sweep etc. I can try to fix this some day to behave more like the other FID parsers. Meanwhile, if you happen to possess the same data already processed to frequency domain, I suggest using it. Anyway, nice to see someone using the parser and good that you found a workaround :) @jjhelmus, feel free to ping me if new issues considering the parser emerge!

Samipsap35, 2020-04-20:
can you please help me, why unit conversion is not possible with bruker data. if  it is  possible how we can do it
import nmrglue as ng
dic, data = ng.bruker.read('1')
udic = ng.bruker.guess_udic(dic, data)
uc = ng.fileiobase.uc_from_udic(udic)
ppm_scale = uc.ppm_scale()
   uc0 = ng.pipe.make_uc(dic,data,dim=0)

kaustubhmote, 2020-04-23:
Hi @Samipsap35, I think there is a mistake in the last line, you are trying to use the `ng.pipe` module (which is only for nmrpipe format) with bruker data. The following should work for bruker data (this is almost exactly your code, without the call to `ng.pipe`, and some comments):

```
import nmrglue as ng
dic, data = ng.bruker.read('1')
udic = ng.bruker.guess_udic(dic, data)
uc0 = ng.fileiobase.uc_from_udic(udic) # defaults to the last dimension
ppm_scale = uc0.ppm_scale()

# use the two lines below if data is 2D
# uc1 = ng.fileiobase.uc_from_udic(udic, axis=0) 
# ppm_scale1 = uc1.ppm_scale() 

```

closed 2026-07-17 by kaustubhmote