jjhelmus/nmrglue #113: ng.pipe.make_uc() type error

hadwin-357, 2020-02-03:
I am using python 3.7.6 and tried both the nmrglue 0.7 and nmrglue 0.8.dev

I am running to a typeerror when I tried to use ng.pipe.make_uc().

Here is my code:

dic, data = ng.pipe.read("./test.ft")
uc2=ng.pipe.make_uc(dic, data)
fig = plt.figure()
ax = fig.add_subplot(111)

ax.plot(uc2.ppm_scale()[::-1], data[::-1], 'k-')

This gives me the following error message:
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
/usr/local/lib/python3.7/site-packages/numpy/core/function_base.py in linspace(start, stop, num, endpoint, retstep, dtype, axis)
    116     try:
--> 117         num = operator.index(num)
    118     except TypeError:

TypeError: 'float' object cannot be interpreted as an integer

During handling of the above exception, another exception occurred:

TypeError                                 Traceback (most recent call last)
<ipython-input-95-211ff63c7bf7> in <module>
      4 ax = fig.add_subplot(111)
      5 
----> 6 ax.plot(uc2.ppm_scale()[::-1], data[::-1], 'k-')

/usr/local/lib/python3.7/site-packages/nmrglue/fileio/fileiobase.py in ppm_scale(self)
    274         """
    275         x0, x1 = self.ppm_limits()
--> 276         return np.linspace(x0, x1, self._size)
    277 
    278     def hz_limits(self):

<__array_function__ internals> in linspace(*args, **kwargs)

/usr/local/lib/python3.7/site-packages/numpy/core/function_base.py in linspace(start, stop, num, endpoint, retstep, dtype, axis)
    119         raise TypeError(
    120             "object of type {} cannot be safely interpreted as an integer."
--> 121                 .format(type(num)))
    122 
    123     if num < 0:

TypeError: object of type <class 'float'> cannot be safely interpreted as an integer.

I also tried the following code and it works:

# read in the data from a NMRPipe file
dic, data = ng.pipe.read("./test.ft")

## create a unit conversion object for the axis
udic = ng.pipe.guess_udic(dic, data)
uc = ng.fileio.fileiobase.uc_from_udic(udic)

fig = plt.figure()
ax = fig.add_subplot(111)

ax.plot(uc.ppm_scale()[::-1], data[::-1], 'k-')

jjhelmus, 2020-02-05:
@hadwin-357 What version of NumPy do you have installed?  I believe some of the recent releases are more strict about automatic type conversion.

reneeotten, 2020-02-11:
I ran into a similar issue today with NumPy version 1.18.1. The problem is that ```linspace``` does not accept floats anymore for the number of samples to generate (this is the upstream NumPy [commit](https://github.com/numpy/numpy/commit/f4dfe833e3e037bb69113f7250fad3699f918cfc)). Explicitly converting that argument to a ```int``` would work, but perhaps making sure that ```._size``` is an integer makes more sense.

reneeotten, 2020-02-11:
as a side-note there are two additional warnings about code in ```fileio/fileiobase.py```:

```
/opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/nmrglue/fileio/fileiobase.py:388: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if unit is 'ppm':
/opt/local/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/nmrglue/fileio/fileiobase.py:391: SyntaxWarning: "is" with a literal. Did you mean "=="?
  elif unit is 'hz':
```

If you want I can make a PR to fix these at the same time.

jjhelmus, 2020-02-22:
@reneeotten Thanks for looking into this issue.  I agree, the size argument should be cast to an integer.  I've implemented this and changed to string comparison to use an equality in #114.

In the future feel free to submit PRs for issues like these, I'd be happy to review them.

closed 2020-02-22 by jjhelmus

reneeotten, 2020-02-24:
thanks @jjhelmus - I will submit a PR in the future. Are you planning to tag/release a new version soon-ish?

jjhelmus, 2020-03-04:
> Are you planning to tag/release a new version soon-ish?

It is about time for a release.  I'll tag one later this week.

commit f02a50a4 (2020-02-22): cast attrs of unit_conversion to appropiate type

Convert the input to the unit_conversion class to an appropiate type.
NumPy 1.18 is stricter about the type of input that can be gives to
certain functions, for example linspace. This makes certain the the
appropiate types are used in these calls.

closes #113