jjhelmus/nmrglue #115: Reading bruker fid file TypeError

Lee4396, 2020-02-29:
When I tried to read raw fid file I got this weird type error

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-35-71d0b4f7753d> in <module>
----> 1 dic,data = ng.bruker.read(dir='./Lee_NMR/S_20190122_00439/1/', bin_file='fid')

c:\users\zwqy\appdata\local\programs\python\python38\lib\site-packages\nmrglue\fileio\bruker.py in read(dir, bin_file, acqus_files, pprog_file, shape, cplex, big, read_pulseprogram, read_acqus)
    325         standard files.
    326     read_procs : bool, optional
--> 327         True to read procs files(s), False prevents reading.
    328 
    329     Returns

c:\users\zwqy\appdata\local\programs\python\python38\lib\site-packages\nmrglue\fileio\bruker.py in read_binary(filename, shape, cplex, big)
    994 
    995     # extract data type from dictionary
--> 996     try:
    997         dtypa = int(dic["acqus"]["DTYPA"])
    998     except KeyError:

TypeError: 'float' object cannot be interpreted as an integer

No clue what's going on here

kaustubhmote, 2020-03-02:
I can reproduce this with version 0.5, but seems like this issue was fixed in version 0.6. This is an error associated with non-integer division in calculating the number of data points. It got taken care of in a rehaul of the `bruker.read` function (offending line 397 in `bruker.py` in v0.5). Previous versions of numpy allowed this, but not anymore. If you are using a version older than 0.6, the best way to fix this is to update to the latest version of nmrglue. 

`pip install nmrglue` should update you to version 0.7, and `pip install git+https://github.com/jjhelmus/nmrglue.git` should update you to the current development version.

Lee4396, 2020-03-02:
True, I updated to 0.7 and fixed the problem. Pre-assigning a value to shape also fixes the problem.

closed 2020-03-02 by Lee4396