jjhelmus/nmrglue #116: UnicodeDecodeError when opening up UCSF files

ostannick, 2020-03-07:
Trying to open up a 3D spectrum (UCSF)

```
>>> import nmrglue as ng
>>> ng.read('data/HNCO.ucsf')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'nmrglue' has no attribute 'read'
>>> dic,data = ng.pipe.read("data/HNCO.ucsf")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\kaneo\AppData\Local\Programs\Python\Python38-32\lib\site-packages\nmrglue\fileio\pipe.py", line 525, in read
    dic = fdata2dic(fdata)
  File "C:\Users\kaneo\AppData\Local\Programs\Python\Python38-32\lib\site-packages\nmrglue\fileio\pipe.py", line 1540, in fdata2dic
    dic["FDSRCNAME"] = _unpack_str('16s', fdata[286:290])
  File "C:\Users\kaneo\AppData\Local\Programs\Python\Python38-32\lib\site-packages\nmrglue\fileio\pipe.py", line 1533, in _unpack_str
    return struct.unpack(fmt, d)[0].decode().strip('\x00')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc8 in position 0: invalid continuation byte
```

Any ideas how to fix this?

JLVarjo, 2020-03-09:
There is an extended ASCII character in your file somewhere (apparently È). As a quick fix try to replace that. In code this can be fixed by augmenting all non-binary file opens and relevant decode calls with 
``
encoding="utf-8", errors="replace"
``
Funny coincidence, I just encountered the same issue and fixed these for all fileio code, so there will be pull request soon :)

kaustubhmote, 2020-03-09:
@ostannick, seems like you are trying to read a sparky file in nmrglue, but using the `pipe.read` function, which will work only for nmrpipe files. You should instead use:

    >>> dic, data = ng.sparky.read("data/HNCO.ucsf")

ostannick, 2020-03-09:
Thank you both, apologies for my idiocy!

closed 2020-03-09 by ostannick