jjhelmus/nmrglue #117: Reading Magritek Jcamp

dwhitaker0, 2020-03-18:
Hi, 

I have a jcamp file from magritek, looks to be similar to bruker. 

I can read and plot using this script but the ppm axis isn't right, any help would be appreciated. 

Thanks, Daz

```
import matplotlib.pyplot as plt
import numpy as np
import nmrglue as ng

# read in the data
fname = "nmr_fid.dx"
dic, raw_data = ng.jcampdx.read(fname)

# create complex array
npoints = int(dic["$TD"][0])
data = np.empty((npoints, ), dtype='complex128')
data.real = raw_data[0][:]
data.imag = raw_data[1][:]

# process
data = ng.proc_base.zf_size(data, npoints)
data = ng.proc_base.fft(data)
data = ng.proc_base.di(data)

# determine the ppm scale
udic = ng.jcampdx.guess_udic(dic, data)
udic[0]['car'] = 0.0
uc = ng.fileiobase.uc_from_udic(udic)
ppm_scale = uc.ppm_scale()

# plot
fig = plt.figure()
ax = fig.add_subplot(111)
ax.plot(data)
plt.show()
```

[nmr_fid.zip](https://github.com/jjhelmus/nmrglue/files/4348588/nmr_fid.zip)

kaustubhmote, 2020-03-19:
Hi, you can try plotting with `ax.plot(ppm_scale, data)` and see if that is what you want. However, some of the parameters that `guess_udic` is reading in do not seem correct to me.  The problem seems to be that the information in the jcampdx corresponds to time domain data, but the data itself is frequency domain. So, the acquisition time is being read in as the spectral width, which is obviously incorrect. This is possibly a bug, but more likely is being caused by differences how things are saved in jcampdx formats used by different vendors.

My suggestion would be to set the `udic` parameters manually to those you know to be correct. At the very least, `udic[0]["sw"]` and `udic[0]["car"]` should be set correctly in this dataset, and `udic[0]["time"] = False` and `udic[0]["freq"] = True` for completeness.

dwhitaker0, 2020-03-19:
Hi, 

Thanks for that - I did actually do `ax.plot(ppm_scale, data)` in the first place but it didn't work. Even with setting those values manually (I think!) it doesn't come out with quite the right range. Its close but still off. 

```
udic = ng.jcampdx.guess_udic(dic, data)
udic[0]['car'] = 61
udic[0]["sw"] = 5000
udic[0]["time"] = False 
udic[0]["freq"] = True
uc = ng.fileiobase.uc_from_udic(udic)
ppm_scale = uc.ppm_scale()
```
Is there any way to read in the processed *.1d file that was in the zip, that might be the easier option?

Thanks

kaustubhmote, 2020-03-19:
You can read the processed data from spectrum.1d by the following code:

    data = np.fromfile("spectrum.1d", "<f")[::-1]
    data = data[1:131072:2] + 1j*data[0:131072:2]

Here, I am trimming it to the data part of the file. However, the processing parameters, which will be given in the initial part of the file, cannot be read off without knowing the how they are stored. For example, you can look at the `fdata2dic` function in `fileio/pipe.py`. Unfortunately, I cannot find any specification file for Magritek data. 

Having said that, l still feel that the carrier frequency that you are using is probably incorrect. By my calculations, it should be close to `291.6`. The following seems to work (at least the peaks are within 0 to 10ppm):

    udic = ng.jcampdx.guess_udic(dic, data)
    udic[0]["car"] = (float(dic["$BF1"][0]) - float(dic["$SF"][0])) * 1e6 # 291.6787706013224
    udic[0]["sw"] = float(dic["$SF"][0])*float(dic["$SW"][0]) # 61.6425172559393*81.112850
    udic[0]["time"] = False 
    udic[0]["freq"] = True
    uc = ng.fileiobase.uc_from_udic(udic)
    ppm_scale = uc.ppm_scale()

dwhitaker0, 2020-03-19:
Ah, that's excellent thanks. 

It does seem to have been a brain fail on my part in the calculations. The ppm axis does now pretty much match (within a rounding error) the axis the mnova gives. 

Much appreciated, I'll close the issue.

closed 2020-03-19 by dwhitaker0