jjhelmus/nmrglue #121: Problems with ppm_scale()

AnsgarSiemer, 2020-07-23:
I recently encountered an issue with plotting simple 1D spectra, after similar scripts have worked for years:

```
TypeError                                 Traceback (most recent call last)
/usr/lib/python3/dist-packages/numpy/core/function_base.py in linspace(start, stop, num, endpoint, retstep, dtype, axis)
    116     try:
--> 117         num = operator.index(num)
    118     except TypeError:

TypeError: 'float' object cannot be interpreted as an integer

During handling of the above exception, another exception occurred:

TypeError                                 Traceback (most recent call last)
<ipython-input-4-48206da35a06> in <module>
     19 fig = plt.figure(figsize=(3.5, 2.5))
     20 ax = fig.add_subplot(111)
---> 21 y=uc.ppm_scale()
     22 #ax.plot(uc.ppm_scale(), data, 'k')
     23 #ax.plot(uc.ppm_scale(), dataII, 'deeppink')

/usr/local/lib/python3.8/dist-packages/nmrglue/fileio/fileiobase.py in ppm_scale(self)
    274         """
    275         x0, x1 = self.ppm_limits()
--> 276         return np.linspace(x0, x1, self._size)
    277 
    278     def hz_limits(self):

<__array_function__ internals> in linspace(*args, **kwargs)

/usr/lib/python3/dist-packages/numpy/core/function_base.py in linspace(start, stop, num, endpoint, retstep, dtype, axis)
    117         num = operator.index(num)
    118     except TypeError:
--> 119         raise TypeError(
    120             "object of type {} cannot be safely interpreted as an integer."
    121                 .format(type(num)))

TypeError: object of type <class 'float'> cannot be safely interpreted as an integer.
```

I was able to fix it by changing 
`return np.linspace(x0, x1, self._size)`
to 
`return np.linspace(x0, x1, int(self._size))`
in line 276 of fileiobase.py. 

This is probably not the best solution. I am running Debian unstable with python 3.8 but I could reproduce the error using miniconda and python 3.6.

Nmrglue is a great module. Thank you for writing and maintaining it.

kaustubhmote, 2020-07-26:
@AnsgarSiemer, Thanks for pointing this out. #114 should have taken care of this error, as well as the related issue #113. This was done by fixing the type of the `_size` attribute to `int`, as you correctly pointed out. I notice that this was just before the v0.7 release, so this fix seems to be currently present only on the `master` branch. If you are using pip to install nmrglue, `python -m pip install git+https://github.com/jjhelmus/nmrglue.git` should get you the latest development version. Let us know if that fixes it.

AnsgarSiemer, 2020-07-27:
@kaustubhmote Thank you. I installed the master branch and the problem went away.

closed 2022-07-22 by kaustubhmote