jjhelmus/nmrglue #122: Tutorial in Colabs

Darren01, 2020-07-31:
Missing File test.ft

I'm working my way through your [tutorial](https://nmrglue.readthedocs.io/en/latest/tutorial.html) in Colabs.

I'm at the section [using-matplotlib-to-create-figures](https://nmrglue.readthedocs.io/en/latest/tutorial.html#using-matplotlib-to-create-figures) but when I try to implement the following,

>>> import pylab
>>> dic,data = ng.pipe.read("test.ft")
>>> pylab.plot(data)
[<matplotlib.lines.Line2D object at 0x8754fd0>]
>>> pylab.savefig("plot_1d.png")

I cannot find the test.ft file.

kaustubhmote, 2020-08-04:
Hi @Darren01 ,
All the datasets associated with the tutorial/testing is available at [Google Code Archive](https://code.google.com/archive/p/nmrglue/downloads). If you download it and follow the instructions [here](https://nmrglue.readthedocs.io/en/latest/devel/index.html) for setting up the directories, you should be able to run this code. 

However, this does seems like a rather long download/process to just test out reading and plotting. One thing you can do is replace `test.ft` with any NMR dataset that you might have. Just replace the `ng.pipe.read` with the command for the dataset type (`ng.bruker.read`, `ng.jcamdx.read`, etc). Worst case, you can copy the data from #112. If you have any problems, please seel free to comment back here.

kaustubhmote, 2020-08-25:
You can also use this file: `nmrglue/tests/pipe_proc_tests/1D_freq_real.dat`, which is included in the github repository.

`dic, data = ng.pipe.read("1D_freq_real.dat")` should work.

Darren01, 2020-08-25:
Thanks for your help, @kaustubhmote .  I am trying to work through the tutorial, but on colab.  Here's what I have to date, https://colab.research.google.com/drive/1AJk9x4HQKGff6ufHNP6uMpi8PmW8onsK?usp=sharing 
I've used some of my own NMR data to create a the image, below but ...

[image]

... I was hoping to follow the tutorial directly but within colabs.  In this way, I was hoping to show how a potential user could operate the software package within colabs.

All the best
Darren.

closed 2022-07-22 by kaustubhmote