jjhelmus/nmrglue #123: Reading tables from tecmag file

wired4life4, 2020-09-04:
Hello,

I am trying to read some custom tables in a .tnt file and could not find any way to do this. I am only away of the one command, 
`dic,data = nmrglue.fileio.tecmag.read(path)`
to read the raw data and parameter dictionary.  I read through the tutorial and documentation and saw that the [chatcannon/pytnt ](https://github.com/chatcannon/pytnt)project has the ability to read delay tables. Is this capability also in nmrglue? Thank you :)

chatcannon, 2020-09-06:
I could try adding the delay tables code to nmrglue. Presumably the delay tables information would go in the parameter dictionary - do you have any preferences as to how it should be structured?

kaustubhmote, 2020-09-07:
@wired4life4, nmrglue currrently does not have the ability to read this, but @chatcannon, It would be great to have this in nmrglue. I think this can be done in a way similar to that in `pytnt`: a dictionary accessible from the main dictionary by a keyword like `delay` might work nicely (I guess it this would be identical to the `DELAY` attribute in `TNTfile` class in `pytnt`).  Also, since I am not too familiar with tecmag datasets, it would be nice to have a test added to verify this functionality. Will the current test data for tecmag suffice for this?

wired4life4, 2020-09-08:
There is one sample file I found in /TNMR/setup/sweep.tnt that does have delay tables, sweep tables, and shape tables, however, the naming looks different than the default when creating them manually.

robbyblum, 2020-09-08:
I am also interested in this feature, because I work with a lot of Tecmag data sets. I don't know how to find the current Tecmag test data (apparently called `LiCl_ref1.tnt`), though; it doesn't seem to be in the old Google Code archive of test data from version 0.4, nor is it in the github data folders. If that's just a simple 1D experiment, though, I strongly doubt it will have _any_ delay tables, of any name scheme.

The name scheme issue is a thorny problem, as @wired4life4 has already found. TNMR lets you name tables quite freely. In the `.tnt` file, the table is stored with a 4-byte integer saying how long the name is, then the name, then a 4-byte integer saying how long the table contents are, then the contents, and then other parameters afterward. As far as I can tell, there isn't a flag that says "TABLES START HERE", they just start right after the last "sequence events" section of the file (hopefully this is always true). So if your tables aren't named according to a formula like that assumed by `pytnt`, I don't know how you would programmatically locate them in the file. In my practice, I pretty much always use descriptive tables names instead of the defaults, and I'm sure I'm not the only one who does this...

kaustubhmote, 2020-09-09:
@robbyblum, the Tecmag test dataset is in the v0.5 test dataset, located [here](https://drive.google.com/folderview?id=0BzCZObue1SeERk52WmJfSVdVZWc&usp=sharing). 

May be a good way of doing this would be adding a separate `guess_tables` function to `tecmag.py`. All the logic for this can go in there, possibly with a optional arguments for the (non default) names of the tables, which the function can search for in addition to the default names. If all works out OK with this function, we can add an option in the main `read` function which simply calls it.