jjhelmus/nmrglue #128: Proposal: release version 0.8.0 and drop support for Python 2.7 and Python < 3.6

jjhelmus, 2020-10-20:
It has been a while since the last release of nmrglue.  Version 0.7.0 was release in Feb, 2019.  There was been a number of [changes since that release](https://github.com/jjhelmus/nmrglue/compare/v0.7...master) and I think it would be good to make a 0.8.0 release.  

Python 2.7 reached it's end of life at the beginning of this year, 2020, and is no longer supported by the CPython core developers.  Version of Python < 3.6 have [also reached end-of-life](https://devguide.python.org/#status-of-python-branches).  The Scientific Python/PyData community has also stopped supporting Python 2.7 and [older version of Python 3](https://numpy.org/neps/nep-0029-deprecation_policy.html#support-table).  [NEP 29](https://numpy.org/neps/nep-0029-deprecation_policy.html#support-table) in particular recommends that the community support Python>=3.7 as of Jun 23, 2020.  

Given this I am proposing on making a 0.8.0 release next week with the expectation that this will be the last release to support Python 2.7 and Python <3.6.  Note that NEP 29 would recommend Python <3.7 but supporting 3.6 is low effort. After the release Python 2 code can be remove from nmrglue and newer Python 3 syntax can be added.  

If anyone has concerns about this it would be good to voice them here.  If there are no concerns I will cut the 0.8.0 release on Tuesday Oct 27.

kaustubhmote, 2020-10-21:
This sounds great. I was hoping to finish adding documentation for new features and changes before v0.8, but it seems unlikely to be be completed soon. One thing that I think should be added is a changelog. I don't think we have this for any of the previous releases. You have any suggestions as to where this should go?

jjhelmus, 2020-10-21:
I agree that is it time to keep a changelog, a `CHANGELOG.txt` file in the root directory seems like a reasonable place for this.  I will put one together shortly.

commit 79494441 (2020-10-29): remove support for Python 2.7 and <3.6

nmrglue requires Python 3.6 or newer.

closes #128
closed 2020-10-29 by jjhelmus