jjhelmus/nmrglue #135: exception when reading Varian Agilent data

andrealorenzon, 2020-12-03:
When using nmrglue.fileio.varian.read() on the folder of Varian Agilent data, I get the following error.

```
ValueError   Traceback (most recent call last)
<ipython-input-88-89a930f3e363> in <module>
----> 1 dic, data = ng.fileio.varian.read('testSpectra')
[...]
ValueError: could not broadcast input array from shape (3192) into shape (4096)
```

Can anyone point me to the right direction to debug this?

kaustubhmote, 2020-12-04:
Hi @andrealorenzon,
Can you share the complete error trace so we can see which line in the given function is giving the error?

andrealorenzon, 2020-12-09:
hi @kaustubhmote .

This is the complete stacktrace:

[image]

testSpectra is a local folder where I have the 4 files: fid, log, procpar and text.

kaustubhmote, 2020-12-10:
It seems that the data parameters being read in from `procpar` are somehow inconsistent with the actual data in `fid`. In particular, the parameters `np` from procpar seems to be off. There are a couple of things you can try:

1. Try reading in the procpar using `dic = ng.varian.read_procpar("testSpectra/procpar")` and see whether all the size parameters are consistent with what you actually expect.
2. Try passing the `as_2d=True` argument to ng.varian.read("testSpectra") (if this is a 2d)

If none of these work, I think I might need to see the actual data to debug the problem. Will it be possible for you to share `fid` and `procpar`?

andrealorenzon, 2021-02-26:
sorry for the late reply: sorry, but I'm not allowed to post the data. Thank you for your assistance, anyway.

closed 2021-02-26 by andrealorenzon