jjhelmus/nmrglue #149: Echo anti-echo aquisition

AJAdkns, 2021-06-08:
I am trying to process data in which the echo anti-echo (or rance-kay) method needs to be applied, but I do not see that this is a feature of nmrglue. I am attempting to do this by manipulating the data manually. I have read that the data needs to be separated into the echo or N-type spectrum and the anti-echo or P-type spectrum, but how would I actually go about obtaining or separating these two spectra within the the numpy array in nmrglue? Any help is much appreciated.

kaustubhmote, 2021-06-09:
If the data are arranged in as "one fid per row", you can just separate out using numpy's array-slicing syntax:

```python
dataP, dataN = data[0::2], data[1::2]
``` 

If you need to convert the data into a "one row per fid" format, you can reshape using the value of the number of points in the direct dimension. For Bruker datasets, this would look like this:

```python

dic, data = ng.bruker.read(filename)
data = data.reshape(-1, dic["acqus"]["TD"])

```

You should also look at the function given here: [nmrgle/process/nmrtxt/rance_kay.py](https://github.com/jjhelmus/nmrglue/blob/master/nmrglue/process/nmrtxt/rance_kay.py) to see how to do the reshuffling. This function is currently not available directly from nmrglue.

AJAdkns, 2021-06-09:
Thank you so much, this will be a lot of help! Just curious, my dataset is in the Varian format, so would it be different than this method with Brucker data or is it relatively the same?

AJAdkns, 2021-06-09:
> If the data are arranged as "one fid per row", you can separate them using Numpy's array-slicing syntax:
> 
> ```python
> dataP, dataN = data[0::2], data[1::2]
> ```
> 
> If you need to convert the data into a "one row per fid" format, you can reshape using the value of the number of points in the direct dimension. For Bruker datasets, this would look like this:
> 
> ```python
> dic, data = ng.bruker.read(filename)
> data = data.reshape(-1, dic["acqus"]["TD"])
> ```
> 
> You should also look at the function given here: [nmrgle/process/nmrtxt/rance_kay.py](https://github.com/jjhelmus/nmrglue/blob/master/nmrglue/process/nmrtxt/rance_kay.py) to see how to do the reshuffling. This function is currently not available directly from nmrglue.

kaustubhmote, 2021-06-10:
The splitting part should be identical. The reading part will be similar except for using the function in the `agilent` module, and the reshaping part will only differ in the parameter to be looked up from the dictionary.

```python
dic, data = ng.agilent.read(filepath)
data = data.reshape(-1, dic["np"])
dataP, dataN = data[0::2], data[1::2]
```

kaustubhmote, 2021-06-11:
I think the P and N datasets should be added while setting `shuffle[2*i+1]` instead of being subtracted. 

Also, I think you are basing this on the function that is appropriate for the way Bruker stores the data. I think the way Varian does this is slightly different. I don't have a handy Varian dataset that I can check this against, but since you seem to have nmrpipe, I suggest looking up the `var_ranceY.M` macros in `nmrpipe/nmrtxt` and comparing that with the `bruk_ranceY.M` in the same folder to see the difference.

AJAdkns, 2021-06-18:
Just a follow up on this issue, I looked at the code for the var_ranceY.M macro in NMRPipe and it gave me a little bit of insight into my own goal. 

I figured out that keeping the data in its original shape instead of reshaping and reshuffling the data simply by stacking the two sets next to each other yields a result closer to what I am seeking.

Here is what this method yields:
[image]

And here is what it should look like:
[image]

And here is the code (for the first version):
```python
vdic, vdata = ng.varian.read('EJS218_0mM_15NHSQC_20C_112018.fid', torder='f', as_2d=True)
dataP, dataN = vdata[0::2], vdata[1::2] 

shuffle = np.empty(vdata.shape, vdata.dtype)
x = int(0.5*vdata.shape[0])
for i in range(0, x):
    shuffle[i] =  1.*(dataP[i].real - dataN[i].real) + 1.*(dataP[i].imag - dataN[i].imag)*1j
    shuffle[i+x] = 1.*(dataP[i].imag + dataN[i].imag) + 1.*(dataP[i].real + dataN[i].real)*1j

C = ng.convert.converter()
C.from_varian(vdic, shuffle)
dic, data = C.to_pipe()
```

Some differences to note, the spectrum seems to be similar but with extra points, and it is flipped about the y-axis. Also, the scales on the axes are way off from the actual chemical shift values. I'm assuming that there is one additional bit of processing that I need but I'm not sure what it is.

kaustubhmote, 2021-06-21:
Not quite sure what might be missing here, but if it is possible for you to share the dataset, I'll be happy to take a look.

AJAdkns, 2021-06-21:
Here is the dataset. Thank you so much for your help once again.

[EJS218_0mM_15NHSQC_20C_112018.fid.zip](https://github.com/jjhelmus/nmrglue/files/6686933/EJS218_0mM_15NHSQC_20C_112018.fid.zip)

AJAdkns, 2021-06-23:
Thank you so much! I suspected I was doing the interleaving wrong, I just had slightly better results from stacking the data. 

The only thing is that when I process and plot the data in exactly the same way, the scales of the axes are still off. For example, the 15N axis is on the order of 0.5-1.5 when it should be 105-135. Would this also be a problem with the conversion?

closed 2021-06-24 by AJAdkns