jjhelmus/nmrglue #150: CSDM conversion?

mccarthy677, 2021-07-21:
Hello!

I'm helping to develop the mrsimulator library (see links below), but it requires CSDM format data for fitting.  I've been using nmrglue to load vendor data into python and manually building a CSDM data object using csdmpy, but a function in nmrglue would make it a lot easier.  I think a to_csdm function in the converter object would work perfectly for this.  I'm willing to work on this and contribute with a pull request if this addition sounds good to you!

Thanks!

mrsimulator library:
https://mrsimulator.readthedocs.io/en/stable/

CSDM data:
https://journals.plos.org/plosone/article?id=10.1371/journal.pone.0225953

csdmpy library:
https://csdmpy.readthedocs.io/en/stable/

kaustubhmote, 2021-08-09:
This looks like a good proposal, and it would be great to have this conversion in nmrglue. But,  having functions to convert from all different formats to `csdm` seems to be a lot of work to take on. Perhaps converting internally to a `universal_dictionary` and then to `csdm` might be a good solution here?  If this is what you have in mind, I would be happy to review any PR from this.

mccarthy677, 2021-08-09:
Yes, that's what I was thinking!  I just created  PR #152 that adds this functionality.  Thanks!