jjhelmus/nmrglue #166: nmrglue  import fails with scipy=1.8

fernandezc, 2022-02-06:
Works well with scipy=1.7.3 but cannot be imported with scipy=1.8.

error:   ImportError: cannot import name '_check_func' from 'scipy.optimize.minpack' 

(nmrglue version 0.9dev0)

```python 
Python 3.9.9 | packaged by conda-forge | (main, Dec 20 2021, 02:38:53)
Type 'copyright', 'credits' or 'license' for more information
IPython 8.0.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import nmrglue
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
Input In [1], in <module>
----> 1 import nmrglue

File ~/opt/miniconda3/envs/scpy_pip/lib/python3.9/site-packages/nmrglue/__init__.py:4, in <module>
      2 from .process import *
      3 from .util import *
----> 4 from .analysis import *
      6 __version__ = '0.8.2-dev'

File ~/opt/miniconda3/envs/scpy_pip/lib/python3.9/site-packages/nmrglue/analysis/__init__.py:2, in <module>
      1 from . import analysisbase
----> 2 from . import leastsqbound
      3 from . import segmentation
      4 from . import peakpick

File ~/opt/miniconda3/envs/scpy_pip/lib/python3.9/site-packages/nmrglue/analysis/leastsqbound.py:7, in <module>
      5 from numpy import array, take, eye, triu, transpose, dot
      6 from numpy import empty_like, sqrt, cos, sin, arcsin
----> 7 from scipy.optimize.minpack import _check_func
      8 from scipy.optimize import _minpack, leastsq
     11 def _internal2external_grad(xi, bounds):

ImportError: cannot import name '_check_func' from 'scipy.optimize.minpack' (/Users/christian/opt/miniconda3/envs/scpy_pip/lib/python3.9/site-packages/scipy/optimize/minpack.py)

In [2]:

```

kaustubhmote, 2022-02-07:
This should be fixed in the latest master branch. Thanks for opening the issue!

balling-dev, 2022-03-02:
Thanks for fixing @kaustubhmote - if we could get this fix into a (patch) release on pypi.org it would be awesome :pray:

jjhelmus, 2022-04-20:
nmrglue 0.9 has been tagged and released on PyPI.  This version includes the fix in #167 that allows nmrglue to be imported in environments with scipy >=1.8.

closed 2022-04-20 by jjhelmus