jjhelmus/nmrglue #170: Installation fails with setuptools v61.2.0

ptrxyz, 2022-04-18:
I recently upgraded setuptools to v61.2.0 (from 58.0.4) and the installation does now fail. The log is super long, but should be reproducible easily in a venv. Here is the last lines of the error in case it helps:

```
...
File "/anaconda3/envs/chem-spectra/lib/python3.7/site-packages/setuptools/_vendor/importlib_metadata/__init__.py", line 886, in find_distributions
         found = self._search_paths(context.name, context.path)
     AttributeError: 'str' object has no attribute 'name'
```

kaustubhmote, 2022-04-20:
I am unable to reproduce this with a clean conda env with python 3.7 (and 3.8/3.9), setuptools 61.2, and the current master branch of nmrglue. Can you provide details about the nmrglue version you are using?

ptrxyz, 2022-04-20:
Yep, I'll have a look and provide more info.