jjhelmus/nmrglue #174: integrate_2d: how to get limits.in?

18270020070, 2022-07-22:
The module code directly reads the limits.in file, but I wonder how the file is obtained, such as  x0, y0, x1, y1 in the file. Whether according to the spectral peak Add or subtract a certain value from the xy coordinate value to get? Thanks a lot for your answer.

kaustubhmote, 2022-07-22:
In most of the cases, you will need to do this manually by looking at the peak position/shape. Once the peaks are picked, you can potentially automate this using the `unit_conversion` object, but you will  should plot and see whether the limits you chose are appropriate. For example, you can use something like:

```python
# generate the universal dictionary (udic) depending on your data format

# generate unit conversion object
uc = {i: ng.fileiobase.uc_from_udic(udic, dim=i) for i in (0, 1)}

# select a 1 ppm box around a 2d peak at the position (x=50ppm, y=55ppm)
x, y = 50, 55
delta = 1

x0, x1 = uc[1].i(x - delta, "ppm"), uc[1].i(x + delta, "ppm")
y0, y1 = uc[0].i(y - delta, "ppm"), uc[0].i(y + delta, "ppm")

```

18270020070, 2022-07-24:
> In most of the cases, you will need to do this manually by looking at the peak position/shape. Once the peaks are picked, you can potentially automate this using the `unit_conversion` object, but [...]

Thank you very much for your answer, I think I have an idea.