jjhelmus/nmrglue #175: Problem with 1D plotting - maybe

mrbatchelor, 2022-10-10:
Maybe an issue with my python (on an old Mac, running High Sierra) but I'm trying to make 2D plots with line scans across, something I've used nmrglue before to do. e.g. using the example data and the plot_2d_pipe_spectrum.py or plot_spectrum.py scripts I get errors:

(base) XXX-MacBook-Pro:s4_2d_plotting fbsmba$ python plot_2d_pipe_spectrum.py 
Traceback (most recent call last):
  File "plot_2d_pipe_spectrum.py", line 21, in <module>
    x = uc_13c.ppm_scale()
  File "/Users/XXXX/anaconda3/lib/python3.6/site-packages/nmrglue/fileio/fileiobase.py", line 276, in ppm_scale
    return np.linspace(x0, x1, self._size)
  File "<__array_function__ internals>", line 6, in linspace
  File "/Users/XXXX/anaconda3/lib/python3.6/site-packages/numpy/core/function_base.py", line 113, in linspace
    num = operator.index(num)
TypeError: 'float' object cannot be interpreted as an integer

Apologies if this is a daft request - I'm not great with python! If you have suggestions that'd be great.

jjhelmus, 2022-10-10:
Can you check the version of nmrglue you have installed and if necessary upgrade to the latest, 0.9,  release?  If you use pip this can be done using `python -m pip install --upgrade nmrglue`

This issue was fixed in #114 and is included in the 0.9 release of nmrglue.

jjhelmus, 2022-10-10:
Just re-checked the details in #114 and the fix is also included in version 0.8.  Version 0.7 and before will contain the bug.

closed 2022-10-11 by jjhelmus