jjhelmus/nmrglue #188: nmrglue fails loading due to scipy.minpack deprecation (1.9.3)

ilsacripante, 2022-12-08:
Dear all,

I started having issues with nmrglue loading after scipy.optimize.minpack deprecation 
the function _check_func is not available anymore

`    import nmrglue as ng

  File ~\Anaconda3\lib\site-packages\nmrglue\__init__.py:4
    from .analysis import *

  File ~\Anaconda3\lib\site-packages\nmrglue\analysis\__init__.py:2
    from . import leastsqbound

  File ~\Anaconda3\lib\site-packages\nmrglue\analysis\leastsqbound.py:7
    from scipy.optimize.minpack import _check_func

ImportError: cannot import name '_check_func' from 'scipy.optimize.minpack' ( REDACTED_PATH\Anaconda3\lib\site-packages\scipy\optimize\minpack.py)`

kaustubhmote, 2022-12-08:
This was fixed in #167. Please see if the lates code from Github's main branch works for you.

jjhelmus, 2023-11-15:
The fix (#167) is available in the 0.10 release of nmrglue available on PyPI.  Closing this issue. Please re-open if this is still an issue.

closed 2023-11-15 by jjhelmus