jjhelmus/nmrglue #190: Does the spectral raw data obtained by 'Echo-Antiecho', 'States', 'TPPI' and 'States-TPPI' require additional manual operations?

18270020070, 2023-02-18:
Whether to perform additional processing on spectral raw data encoded in 'Echo-Antiecho', 'Atates', 'TPPI' and 'States-TPPI' before converting to pipe format? Then perform operations such as zero-filling, Fourier transform, and imaginary part removal on the direct and indirect dimensions. I'm not very professional with spectral raw data, if I can get your help I will be very grateful.

kaustubhmote, 2023-02-20:
`States`, `States-TPPI`, and `TPPI` data wont require any additional processing other than setting the correct value for the `alt` parameter in the `ft` function. Echo-Antiecho will require additional processing, which you can see in #149.

18270020070, 2023-02-20:
Thanks a lot, so is it right to write like below?
         
[image]