jjhelmus/nmrglue #192: ng.analysis.peakpick.pick() generating deprecation warning due to np.float()

weisscharlesj, 2023-04-13:
The np.analysis.peakpick.pick() function is generating a deprecation warning due to the deprecation of the np.float() function. Looking at the release notes, the easiest fix is to change "np.float()" to "float()" which will not modify the original behavior. An excerpt of the warning is below.

~/miniconda3/lib/python3.10/site-packages/nmrglue/analysis/peakpick.py in pack_table(locations, cluster_ids, scales, amps, axis_names)
    327     anames = axis_names[-ndim:]
    328 
--> 329     dt = [(a + "_AXIS", np.float) for a in anames]
    330     rec = np.rec.array(locations, dtype=dt)
    331 

AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations

weisscharlesj, 2023-04-15:
Numpy 1.24 has broken the ng.analysis.peakpick.pick() function for me now.

kaustubhmote, 2023-06-08:
This is fixed by #196. I will merge this PR soon.

jjhelmus, 2023-11-15:
This fix is available in the 0.10 release of nmrglue, available on PyPI.

closed 2023-11-15 by jjhelmus