jjhelmus/nmrglue #195: Issue when trying to convert Varian data to Pipe

SanderBash, 2023-06-08:
When trying to convert Varian data to Pipe, I get the following error:

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[3], line 12
     10 C = ng.convert.converter()
     11 C.from_varian(vdic, vdata)
---> 12 pdic, pdata = C.to_pipe()
     14 udic = ng.varian.guess_udic(vdic, vdata)
     17 adic,adata = ng.pipe_proc.sine(vdic,vdata,-0.7,0.5,2)

File ~\anaconda3\lib\site-packages\nmrglue\fileio\convert.py:330, in converter.to_pipe(self, datetimeobj)
    327 else:
    328     self._odtype = "float32"
--> 330 return dic, self.__returndata()

File ~\anaconda3\lib\site-packages\nmrglue\fileio\convert.py:72, in converter.__returndata(self)
     70 # Return data
     71 if isinstance(self._data, np.ndarray):   # in memory data
---> 72     return self.__procdata()
     74 else:   # return emulated data
     75     iproc = self._iproc

File ~\anaconda3\lib\site-packages\nmrglue\fileio\convert.py:95, in converter.__procdata(self)
     93     for i in range(data.ndim - 1):
     94         s[i] = slice(1, None, 2)
---> 95         data[s] = -data[s]
     96         s[i] = slice(None, None, None)
     98 if "realfactor" in self._iproc:

IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`) and integer or boolean arrays are valid indices

I was using this code snippet:
C = ng.convert.converter()
C.from_varian(vdic, vdata)
pdic, pdata = C.to_pipe()

kaustubhmote, 2023-06-08:
This is fixed by #196. I'll leave the PR open for a day and then merge it if there are no corrections.

jjhelmus, 2023-11-15:
Closing as fix in #196 has been merged.

closed 2023-11-15 by jjhelmus