jjhelmus/nmrglue #199: How to Process and display multiple 1D Bruker data

Ricmelis80, 2023-09-07:
Hi, I tried reading and processing multiple raw 1D  Bruker NMR files by using the follow code : 

"root_folder = 'nmr'
zero_fill_size = 32768
data = [] # <1>
for folder in os.listdir(root_folder):
    dic, fid = ng.fileio.bruker.read(os.path.join(root_folder,folder))
    fid = ng.bruker.remove_digital_filter(dic, fid)
    fid = ng.proc_base.zf_size(fid, zero_fill_size) # <2>
    fid = ng.proc_base.rev(fid) # <3>
    fid = ng.proc_base.fft(fid)
    data.append(fid)
data = np.array(data)"

However, as 1D fids are located in specific subfolders named "1" (i.e the orginal Topspin EXPNO number) , I'm not able to find them with the bruker. read function.  For more clarity I have attached a zip files containing the data structure of the root folder   

[nmr.zip](https://github.com/jjhelmus/nmrglue/files/12547380/nmr.zip)

Any suggestions? Thanks in adavance!

kaustubhmote, 2023-09-08:
`ng.bruker.read` needs to be pointed to the folder where the `ser`/ `fid` and the `acquNs` files exist. So `dic, fid = ng.bruker.read(os.path.join(root_folder, folder, '1'))` should work here. If the numbers are not always '1', you will have to use `os.listdir` one more time to get the subfolder.

Ricmelis80, 2023-09-08:
I see... Thanks a lot Kaustubh R. Mote!

jjhelmus, 2023-11-15:
Closing as this has been answered.
@Ricmelis80 feel free to re-open if you need additional help with this problem.

closed 2023-11-15 by jjhelmus