jjhelmus/nmrglue #200: Port tests away from nose

SoapGentoo, 2023-09-08:
Hi nmrglue,
we noticed that your package still uses `nose`, which has long been abandoned and is unmaintained. Please consider porting to pytest, otherwise we might have to remove it from Gentoo. Our upstream bug: https://bugs.gentoo.org/878725

kaustubhmote, 2023-09-08:
All nmrglue test are compatible with pytest, and are actually tested with pytest. However, there are a few places where nose is imported for things that can be done easily with pytest, or the imported code is not used. This should be a simple fix, and I will mark this as a bug to be fixed at the earliest.

jjhelmus, 2023-11-15:
#201 Removed the use of `nose` for test in favor of `pytest`.

closed 2023-11-15 by jjhelmus