jjhelmus/nmrglue #202: New release

lukasturcani, 2023-09-15:
Hi there! Thank you for the great library! 

Would it be possible to make a new PyPI release? I think the current PyPI version
runs into issues with the latest Python & NumPy versions because `np.float` has been
removed. This has been fixed in the repo but hasnt been updated in the PyPI release.

Thanks!

kaustubhmote, 2023-09-16:
@jjhelmus 
#189 and #201 can also be merged in before v0.10 is released on PyPI.

jjhelmus, 2023-11-15:
A new release is long overdue.  Let me merge a few outstanding PR and then I'll tag a 0.10 version.

jjhelmus, 2023-11-15:
Wheels and a source distribution for the [0.10 release of nmrglue](https://pypi.org/project/nmrglue/0.10/) are now available on PyPI

closed 2023-11-15 by jjhelmus