jjhelmus/nmrglue #214: Help plotting Bruker 2D spectra

ndbryant21, 2023-12-27:
Does anyone have a script for plotting Bruker 2D spectra? I am new to nmrglue and admittedly am not great with python, so if someone has this developed and is willing to share I would greatly appreciate it. If it matters, I have included my experimental details below. Thanks!

Bruker pulse sequence hsqcetgpsip2.2 with the following parameters: spectra width of 12 ppm in the 1H dimension with 1024 data points; spectra width of 220 ppm in the 13C dimension with 256 increments and 32 scans.

kaustubhmote, 2023-12-29:
@ndbryant21 do you want to only plot the processed data, or process it from scratch and then plot it?

ndbryant21, 2023-12-29:
@kaustubhmote For now I would like to plot the processed data. Though I have been generating a few hundred 2D spectra per year and would eventually like to process from scratch.

kaustubhmote, 2023-12-30:
This should work:

```python

# read data
dic, data = ng.bruker.read_pdata("datafolder/1/pdata/1")

# read in axis ppm parameters
udic = ng.bruker.guess_udic(dic, data, strip_fake=True)
uc = {i: ng.fileiobase.uc_from_udic(udic, dim=i) for i in (0, 1)}
axis_limits = [*uc[1].ppm_limits(), *uc[0].ppm_limits()]

# contour levels
levels = [data.max() / 20 * 1.4**i for i in range(10)]  # change as needed

# plot
fig, ax = plt.subplots()
ax.contour(
    data.real, 
    levels=levels, 
    extent=axis_limits, 
    cmap=plt.cm.Greys_r, # change as per your preference 
    linewidths=0.5, # change as per your preference
)

ax.set_xlim(11.5, 6)  # change as needed
ax.set_ylim(135, 105)  # change as needed

plt.show()
```

I will not yet recommend processing multidimensional data with nmrglue if you need to apply linear prediction to the indirect dimensions and nmrglue's linear prediction functions are not very well tested. It would be better process it in nmrpipe or some other software and then use nmrglue to convert the data to numpy arrays for plotting.