jjhelmus/nmrglue #215: Issue with ng.peakpick.pick for "thres" and "thres-fast" algorithms with Numpy ^v1.22

rserial, 2024-01-03:
**Problem:**
When using `ng.peakpick.pick` with the algorithms "thres" and "thres-fast" and the Numpy version ^1.22, the following error occurs:

```
Error:
peakpick.py:369 # find the rectangular region around the segment
peakpick.py:370 region = data[seg_slice]
peakpick.py:371 edge = [s.start for s in seg_slice]
peakpick.py:372 rlocation = [l - s.start for l, s in zip(location, seg_slice)]

IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`), and integer or boolean arrays are valid indices.
```

It appears to be related to the function call `find_all_thres_fast(data, pthres, msep, True)` since it returns a list for `seg_slice`.

**Potential Fix:**
Modifying line 370 in peakpick.py to:

```python
peakpick.py:370 region = data[seg_slice[0]]
```
or adjusting the return of the find_all_thres_fast function accordingly.

Somehow this is not needed for numpy versions below 1.22 but for above versions it becomes a problem.

kaustubhmote, 2024-01-09:
Strange. I'll provisionally mark this as a bug, although it should have been fixed by #196.
I will take a look at this as soon as possible. Just for confirmation, you are using nmrglue version 0.10 that was released a few weeks ago right?

rserial, 2024-01-09:
Hi! thanks for replying, I am using version 0.10, but now i see that issue #197 was included in the release instead of #196 which has the modication needed for the slices in line 370 (i think).

PR #216 (2024-02-07): Peakpick testing
This PR addresses and fixes #215 by making the API for all algorithms work in the same way.  A self-contained test is also included for the `peakpick` module.
closed 2024-02-07 by kaustubhmote