jjhelmus/nmrglue #218: Reading JEOL NMR Data (*.jdf)

jnimoth, 2024-07-05:
I think that nmrglue can currently not directly read JEOL NMR data (jdf files). It would be cool if that could be also implemented, similiarly to Bruker/Varian/etc.

kaustubhmote, 2024-07-05:
Yes. This is one important thing that is currently missing from nmrglue. I have been working on this intermittently, but have not been able to take it anywhere near completeness. It's a bit of grunt work since I don't have the jdf format specification and have to rely on translations from nmrpipe and a few other libraries. If anyone has access to the format specification and can share it here, I'll be happy to take this forward with priority.

kaustubhmote, 2024-07-09:
I have received specifications now. I will work on this in the next few weeks.

jnimoth, 2024-09-20:
Hi,
I was wondering if there is already some progress/update concerning JEOL (jdf) files?

kaustubhmote, 2024-10-26:
My apologies, this is delayed a bit. I have a partially working code, but this will require a bit more work and some testing from my side, which I will not be able to find time for at least a couple more weeks.

kaustubhmote, 2025-04-18:
Closing this, as #228 adds this functionality.

closed 2025-04-18 by kaustubhmote