jjhelmus/nmrglue #220: Request for New PyPI Release

fedejaure, 2024-09-21:
Are you considering publishing a new release on PyPI? There have been some updates since the last release, and it would be great to have them available.

rserial, 2024-10-25:
Hi @jjhelmus, we want to publish a release of [spinsolveproc](https://github.com/rserial/spinsolveproc) on PyPI. The last nmrglue release works great, but we need the latest features in the repo for our library. Since PyPI doesn’t allow direct repo dependencies, could we get a new PyPI release of nmrglue? Thanks a lot!

jjhelmus, 2024-10-29:
A new release would be great. I'll tag a commit and start the upload process.

jjhelmus, 2024-11-02:
nmrglue version 0.11.0 is tagged and files are available on PyPI.

closed 2024-11-02 by jjhelmus

fedejaure, 2024-11-03:
Thanks a lot, @jjhelmus! Really appreciate all the work!