jjhelmus/nmrglue #223: Strange behavior of proc_pipe.tp

djacob65, 2024-11-07:
Hi,

I try to process a JRES spectrum. At the first stage here how I thought to process it :

```
params, data = ng.bruker.read(path)

C = ng.convert.converter()
C.from_bruker(params, data)
dic, data = C.to_pipe()
print('Read: ', data.shape)

dic, data = ng.pipe_proc.sine(dic, data, off=0, end=1, pow=1, c=1.0)
dic, data = ng.pipe_proc.zf(dic, data, zf=1)
dic, data = ng.pipe_proc.rev(dic, data)
print('ZFx2: ', data.shape)

dic, data = ng.pipe_proc.tp(dic, data)
print('T:    ', data.shape)

dic, data = ng.pipe_proc.sine(dic, data, off=0, end=1, pow=1, c=0.5)
dic, data = ng.pipe_proc.zf(dic, data, zf=0, size=64)
print('PAD:  ', data.shape)
dic, data = ng.pipe_proc.zf(dic, data, zf=1)
print('ZFx2: ', data.shape)
 ...
```

which gives:
```
Read:  (40, 4096)
ZFx2:  (40, 8192)
T:     (8192, 20)  <--- Wrong !!
PAD:   (8192, 64)
ZFx2:  (8192, 128)
...
```

So I replaced the call to proc_pipe.tp with:
```
dic, data0 = ng.pipe_proc.tp(dic, data)
data = data.T
dic["FDSLICECOUNT"] = data.shape[0]
dic["FDSIZE"] = data.shape[1]
dic["FDSPECNUM"] = dic["FDSLICECOUNT"]
```

which gives:
```
Read:  (40, 4096)
ZFx2:  (40, 8192)
T:       (8192, 40)  <---  Goog !!
PAD:   (8192, 64)
ZFx2:  (8192, 128)
...
```

Do you have an explanation for this behavior?
Thanks in advance
Sincerely

kaustubhmote, 2024-11-10:
This maybe related to the flags that are set for hypercomplex data during the conversion from Bruker to Pipe formats. Can you share what the outputs of `dic["FDF" + str(int(dic["FDDIMORDER"][0])) + "QUADFLAG"]` and `dic["FDF" + str(int(dic["FDDIMORDER"][1])) + "QUADFLAG"]` after the conversion are?

djacob65, 2024-11-10:
Here it is:

````
params, data = ng.bruker.read(path)

C = ng.convert.converter()
C.from_bruker(params, data)
dic, data = C.to_pipe()
print('Read: ', data.shape)

param1 = "FDF" + str(int(dic["FDDIMORDER"][0])) + "QUADFLAG"
param2 = "FDF" + str(int(dic["FDDIMORDER"][1])) + "QUADFLAG"
print(param1 + '- ' + param2)
print(dic[param1], dic[param2])
print(dic[param1] and dic[param2])
````

gives:
````
Read:  (40, 4096)
FDF2QUADFLAG- FDF1QUADFLAG
0.0 0.0
0.0
````

fernandezc, 2026-09-21:
The `(8192, 20)` shape comes from `FDF1QUADFLAG=0`. As a result, `pipe_proc.tp()` treats the indirect dimension as complex and combines the 40 rows into 20 real/imaginary pairs.

This flag is generated during the Bruker-to-NMRPipe conversion. Could you report:

```python
params["acqu2s"]["FnMODE"]
```

If it is `1` or `2` (QF/magnitude), the converter is incorrectly keeping the indirect dimension as complex. A focused PR with a regression test can then be proposed.